purduesigbots/pros-atom #6: Uncaught TypeError: Cannot read property 'version' of null

PixelToast, 2016-09-23:
Creating a new PROS project failed

**Atom Version**: 1.10.2
**Electron Version**: 0.37.8
**System**: Debian GNU/Linux
**Thrown From**: [pros](https://github.com/purduesigbots/pros-atom) package, v0.3.1

### Stack Trace

Uncaught TypeError: Cannot read property 'version' of null

```
At /home/pixel/.atom/packages/pros/lib/views/new-project/new-project-view.coffee:73

TypeError: Cannot read property 'version' of null
    at NewProjectView.module.exports.NewProjectView.NewProjectView.create (/home/pixel/.atom/packages/pros/lib/views/new-project/new-project-view.coffee:73:76)
    at HTMLButtonElement.<anonymous> (/home/pixel/.atom/packages/pros/lib/views/new-project/new-project-view.coffee:21:32)
```

### Commands

```
  2x -5:27.3.0 PROS:New-Project (atom-text-editor.editor.is-focused)
     -2:00.2.0 settings-view:open (atom-text-editor.editor)
     -1:53.4.0 command-palette:toggle (div.panels-item)
     -1:23.4.0 window:toggle-dev-tools (div.panels-item)
  2x -1:11.1.0 PROS:New-Project (div.panels-item)
```

### Config

```json
{}
```

### Installed Packages

```coffee
# User
atom-beautify, v0.29.12 (inactive)
build, v0.65.0 (active)
busy, v0.7.0 (active)
linter, v1.11.16 (active)
platformio-ide-terminal, v2.2.0 (active)
pros, v0.3.1 (active)
tool-bar, v1.0.1 (active)
tool-bar-main, v0.0.9 (active)
atom-dark-syntax, v0.27.0 (inactive)
atom-dark-ui, v0.52.0 (inactive)
atom-light-syntax, v0.28.0 (inactive)
atom-light-ui, v0.44.0 (inactive)
base16-tomorrow-dark-theme, v1.1.0 (inactive)
base16-tomorrow-light-theme, v1.1.1 (inactive)
one-dark-ui, v1.5.0 (active)
one-light-ui, v1.5.0 (inactive)
one-dark-syntax, v1.3.0 (active)
one-light-syntax, v1.3.0 (inactive)
solarized-dark-syntax, v1.0.2 (inactive)
solarized-light-syntax, v1.0.2 (inactive)
about, v1.5.3 (active)
archive-view, v0.61.1 (active)
autocomplete-atom-api, v0.10.0 (active)
autocomplete-css, v0.11.2 (active)
autocomplete-html, v0.7.2 (active)
autocomplete-plus, v2.31.1 (active)
autocomplete-snippets, v1.11.0 (active)
autoflow, v0.27.0 (inactive)
autosave, v0.23.1 (active)
background-tips, v0.26.1 (active)
bookmarks, v0.41.0 (active)
bracket-matcher, v0.82.1 (active)
command-palette, v0.38.0 (active)
deprecation-cop, v0.54.1 (active)
dev-live-reload, v0.47.0 (active)
encoding-selector, v0.22.0 (active)
exception-reporting, v0.39.0 (active)
find-and-replace, v0.201.0 (inactive)
fuzzy-finder, v1.3.0 (active)
git-diff, v1.1.0 (active)
go-to-line, v0.31.0 (inactive)
grammar-selector, v0.48.1 (active)
image-view, v0.58.2 (active)
incompatible-packages, v0.26.1 (active)
keybinding-resolver, v0.35.0 (active)
line-ending-selector, v0.5.0 (active)
link, v0.31.1 (inactive)
markdown-preview, v0.158.0 (active)
metrics, v0.53.1 (active)
notifications, v0.65.0 (active)
open-on-github, v1.2.0 (inactive)
package-generator, v1.0.0 (inactive)
settings-view, v0.242.2 (active)
snippets, v1.0.2 (active)
spell-check, v0.67.1 (active)
status-bar, v1.4.1 (active)
styleguide, v0.47.0 (active)
symbols-view, v0.113.0 (inactive)
tabs, v0.100.2 (active)
timecop, v0.33.2 (active)
tree-view, v0.208.2 (active)
update-package-dependencies, v0.10.0 (active)
welcome, v0.34.0 (active)
whitespace, v0.33.0 (active)
wrap-guide, v0.38.1 (active)
language-c, v0.52.1 (active)
language-clojure, v0.21.0 (active)
language-coffee-script, v0.47.2 (active)
language-csharp, v0.12.1 (active)
language-css, v0.37.1 (active)
language-gfm, v0.88.0 (active)
language-git, v0.15.0 (active)
language-go, v0.42.1 (active)
language-html, v0.45.1 (active)
language-hyperlink, v0.16.0 (active)
language-java, v0.23.0 (active)
language-javascript, v0.119.0 (active)
language-json, v0.18.2 (active)
language-less, v0.29.5 (active)
language-make, v0.22.2 (active)
language-mustache, v0.13.0 (active)
language-objective-c, v0.15.1 (active)
language-perl, v0.35.0 (active)
language-php, v0.37.2 (active)
language-property-list, v0.8.0 (active)
language-python, v0.45.0 (active)
language-ruby, v0.69.0 (active)
language-ruby-on-rails, v0.25.0 (active)
language-sass, v0.56.0 (active)
language-shellscript, v0.22.4 (active)
language-source, v0.9.0 (active)
language-sql, v0.23.0 (active)
language-text, v0.7.1 (active)
language-todo, v0.28.0 (active)
language-toml, v0.18.0 (active)
language-xml, v0.34.9 (active)
language-yaml, v0.26.0 (active)

# Dev
No dev packages
```

edjubuh, 2016-09-23:
Hey @PixelToast, do you have the PROS CLI ([purduesigbots/pros-cli](https://github.com/purduesigbots/pros-cli)) installed?

PixelToast, 2016-09-23:
I have files in /usr/local/lib/python2.7/dist-packages/proscli but it appears that no binaries were installed, i don't have cx-freeze and pip fails to build it, ugh

jabibi100, 2016-09-23:
An immediate issue i see by your file path is that the CLI requires python 3.5 and is incompatible with python 2.

PixelToast, 2016-09-23:
whoops thats from when i originally tried installing with pip instead of pip3, i do also have /usr/local/lib/python3.5/dist-packages/proscli

PixelToast, 2016-09-23:
after fighting with pip, manually compiling cx-freeze and manually invoking the pros-cli build script i managed to install it

now i have a new issue, the atom plugin is giving me the error "No templates have been downloaded! Use `pros conduct download` to download the latest kernel."

and some odd behavior in the CLI

`pixel@pixel-PC:~$ pros conduct lstemplate
Available kernels:
Version    Depot          Online    Offline
---------  -------------  --------  ---------
2.10.0     pros-mainline  online
2.11.0     pros-mainline  online
pixel@pixel-PC:~$ pros conduct lsdepot
[('pros-mainline', 'github-releases', 'purduesigbots/pros')]
Name           Registrar        Location
-------------  ---------------  ------------------
pros-mainline  github-releases  purduesigbots/pros
pixel@pixel-PC:~$ pros conduct register ~/pros_test
No templates have been downloaded! Use `pros conduct download` to download the latest kernel or specify a kernel manually.
Aborted!
pixel@pixel-PC:~$ pros conduct register ~/pros_test 2.11.0
Registering /home/pixel/pros_test with kernel 2.11.0
`

edjubuh, 2016-09-23:
cx_Freeze is only a requirement to produce frozen binaries.

`pros conduct register ~/pros_test` will look through your downloaded kernels and select the highest versioned template. Since you don't have any downloaded, you get the message that you need to run `pros conduct download`.

`pros conduct register ~/pros_test 2.11.0` will not verify if the kernel version 2.11.0 exists locally - and will just create the `project.pros` with that version accordingly.

PixelToast, 2016-09-23:
yes i ran pros conduct download, here i did it again
```
pixel@pixel-PC:~$ pros conduct download
Fetching online listing to verify available templates.
Resolved kernel latest to kernel 2.11.0
Downloading kernel 2.11.0 from pros-mainline using github-releases
Fetching release on purduesigbots/pros with tag 2.11.0
Downloading kernel-template.zip (v: 2.11.0)  [####################################]  100%
Extracting kernel-template.zip  [####################################]  100%
Template downloaded to /home/pixel/.config/pros/pros-mainline/kernel-2.11.0
To create a new PROS project with this kernel, run `pros conduct new <folder> 2.11.0 pros-mainline`,
or to upgrade an existing project, run `pros conduct upgrade <folder> 2.11.0 pros-mainline
pixel@pixel-PC:~$ ls /home/pixel/.config/pros/pros-mainline/
depot.pros  kernel-2.11.0
pixel@pixel-PC:~$ pros conduct lstemplate
Available kernels:
Version    Depot          Online    Offline
---------  -------------  --------  ---------
2.11.0     pros-mainline  online
2.10.0     pros-mainline  online
```

edjubuh, 2016-09-23:
Try again, I've reuploaded the kernel templates to the release.

PixelToast, 2016-09-23:
Yep i can create a project now

now my only issue is the linter complains https://dl.dropboxusercontent.com/1/view/bbbzwwmml6inq3r/Apps/Shutter/Selection_043.png

other than that it works, hello world builds and runs fine
thanks!

one thing to ask is a user friendly way of telling you that you don't have permission on the serial port
```
  File "/usr/local/lib/python3.5/dist-packages/serial/serialposix.py", line 433, in in_waiting
    s = fcntl.ioctl(self.fd, TIOCINQ, TIOCM_zero_str)
AttributeError: 'Serial' object has no attribute 'fd'
```
would be confusing to most people, and handling of when the port disconnects would help an application using the CLI figure out whats wrong

oddly running pros help crashes
```
pixel@pixel-PC:~/pros_test$ pros help
Traceback (most recent call last):
  File "/usr/local/bin/pros", line 11, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.5/dist-packages/proscli/main.py", line 10, in main
    cli.main(prog_name='pros')
  File "/usr/local/lib/python3.5/dist-packages/click/core.py", line 696, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python3.5/dist-packages/click/core.py", line 1060, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python3.5/dist-packages/click/core.py", line 889, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python3.5/dist-packages/click/core.py", line 534, in invoke
    return callback(*args, **kwargs)
  File "/usr/local/lib/python3.5/dist-packages/click/decorators.py", line 17, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/usr/local/lib/python3.5/dist-packages/proscli/main.py", line 22, in help_cmd
    click.echo(prosconductor.providers.utils.get_all_available_templates())
AttributeError: module 'prosconductor.providers.utils' has no attribute 'get_all_available_templates'
```
but no error when running pros by itself

closed 2016-09-23 by PixelToast