purduesigbots/pros-atom #16: Pros project not being created

kmindspark, 2016-11-20:
#### Expected Behavior:
Project created
#### Actual Behavior:
Failed to create PROS project.
#### Steps to reproduce:

#### System information:
Operating System:  (Windows, OS X, Ubuntu)

Atom Version: (Atom Version)

Plugin Version: (Plugin Version)

#### Additional Information

#### Screenshots

jabibi100, 2016-11-21:
@kmindspark can you please complete the form and provide as many details as possible. Until then we will not be able to proceed in helping you debug this issue.

kmindspark, 2016-11-22:
Hi, I have another thread which details the error message I get just when trying to open atom, before even trying to create a new project. Thanks :)

HotelCalifornia, 2016-11-22:
#17 seems to supercede this issue

closed 2016-11-22 by HotelCalifornia