purduesigbots/pros-atom #17: Uncaught TypeError: Cannot read property '1' of null

kmindspark, 2016-11-20:
[Enter steps to reproduce below:]

1. ...
2. ...

**Atom Version**: 1.12.4
**Electron Version**: 1.3.9
**System**: Mac OS X 10.12
**Thrown From**: [pros](https://github.com/purduesigbots/pros-atom) package, v0.5.4

### Stack Trace

Uncaught TypeError: Cannot read property '1' of null

```
At /Users/kaushikshivakumar/.atom/packages/pros/lib/views/welcome/welcome-view.coffee:75

TypeError: Cannot read property '1' of null
    at /Users/kaushikshivakumar/.atom/packages/pros/lib/views/welcome/welcome-view.coffee:75:98
    at ChildProcess.<anonymous> (/Users/kaushikshivakumar/.atom/packages/pros/lib/terminal-utilities.coffee:29:7)
    at emitTwo (events.js:106:13)
    at ChildProcess.emit (events.js:191:7)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:215:12)
```

### Commands

```
```

### Config

```json
{
  "core": {},
  "pros": {
    "googleAnalytics": {
      "cid": "0e70208f-2cb1-4efe-a3f4-ef5e3ac19646"
    }
  }
}
```

### Installed Packages

```coffee
# User
atom-beautify, v0.29.13 (inactive)
build, v0.67.0 (active)
busy, v0.7.0 (active)
file-icons, v1.7.25 (active)
linter, v1.11.18 (active)
platformio-ide-terminal, v2.2.2 (active)
pros, v0.5.4 (active)
tool-bar, v1.0.1 (active)
tool-bar-main, v0.0.10 (active)
atom-dark-syntax, v0.27.0 (inactive)
atom-dark-ui, v0.52.0 (inactive)
atom-light-syntax, v0.28.0 (inactive)
atom-light-ui, v0.45.0 (inactive)
base16-tomorrow-dark-theme, v1.3.0 (inactive)
base16-tomorrow-light-theme, v1.3.0 (inactive)
one-dark-ui, v1.6.2 (active)
one-light-ui, v1.6.2 (inactive)
one-dark-syntax, v1.5.0 (active)
one-light-syntax, v1.5.0 (inactive)
solarized-dark-syntax, v1.0.5 (inactive)
solarized-light-syntax, v1.0.5 (inactive)
about, v1.7.0 (active)
archive-view, v0.62.0 (active)
autocomplete-atom-api, v0.10.0 (active)
autocomplete-css, v0.13.1 (active)
autocomplete-html, v0.7.2 (active)
autocomplete-plus, v2.31.4 (active)
autocomplete-snippets, v1.11.0 (active)
autoflow, v0.27.0 (inactive)
autosave, v0.23.1 (active)
background-tips, v0.26.1 (active)
bookmarks, v0.42.0 (active)
bracket-matcher, v0.82.2 (active)
command-palette, v0.39.0 (inactive)
deprecation-cop, v0.54.1 (active)
dev-live-reload, v0.47.0 (active)
encoding-selector, v0.22.0 (active)
exception-reporting, v0.40.0 (active)
find-and-replace, v0.202.2 (inactive)
fuzzy-finder, v1.4.0 (active)
git-diff, v1.1.0 (active)
go-to-line, v0.31.0 (inactive)
grammar-selector, v0.48.2 (active)
image-view, v0.60.0 (active)
incompatible-packages, v0.26.1 (active)
keybinding-resolver, v0.35.0 (active)
line-ending-selector, v0.5.0 (active)
link, v0.31.2 (inactive)
markdown-preview, v0.158.8 (active)
metrics, v1.1.2 (active)
notifications, v0.65.1 (active)
open-on-github, v1.2.1 (inactive)
package-generator, v1.0.1 (inactive)
settings-view, v0.243.1 (active)
snippets, v1.0.4 (active)
spell-check, v0.68.4 (active)
status-bar, v1.4.1 (active)
styleguide, v0.47.2 (active)
symbols-view, v0.113.1 (inactive)
tabs, v0.103.0 (active)
timecop, v0.33.2 (active)
tree-view, v0.210.0 (active)
update-package-dependencies, v0.10.0 (active)
welcome, v0.35.1 (active)
whitespace, v0.35.0 (active)
wrap-guide, v0.38.2 (active)
language-c, v0.54.0 (active)
language-clojure, v0.22.1 (active)
language-coffee-script, v0.48.0 (active)
language-csharp, v0.13.0 (active)
language-css, v0.40.1 (active)
language-gfm, v0.88.0 (active)
language-git, v0.15.0 (active)
language-go, v0.43.0 (active)
language-html, v0.46.1 (active)
language-hyperlink, v0.16.1 (active)
language-java, v0.24.0 (active)
language-javascript, v0.122.0 (active)
language-json, v0.18.3 (active)
language-less, v0.29.6 (active)
language-make, v0.22.2 (active)
language-mustache, v0.13.0 (active)
language-objective-c, v0.15.1 (active)
language-perl, v0.37.0 (active)
language-php, v0.37.3 (active)
language-property-list, v0.8.0 (active)
language-python, v0.45.1 (active)
language-ruby, v0.70.2 (active)
language-ruby-on-rails, v0.25.1 (active)
language-sass, v0.57.0 (active)
language-shellscript, v0.23.0 (active)
language-source, v0.9.0 (active)
language-sql, v0.25.0 (active)
language-text, v0.7.1 (active)
language-todo, v0.29.1 (active)
language-toml, v0.18.1 (active)
language-xml, v0.34.12 (active)
language-yaml, v0.27.1 (active)

# Dev
No dev packages
```

HotelCalifornia, 2016-11-20:
Please enter any steps necessary to reproduce this error.

edjubuh, 2016-11-21:
@kmindspark, how did you install PROS for Atom?

kmindspark, 2016-11-22:
Hi, I installed PROS through the website which came with atom, and then had to update the version of atom that came with pros.

I also tried downloading Atom separately and then adding the pros package, but the same error was given.

edjubuh, 2016-11-22:
Can you open a terminal and run `pros`? Please share the result of running that command here.

kmindspark, 2016-11-22:
-bash: pros: command not found

HotelCalifornia, 2016-11-22:
Okay, can you run the command `echo $PATH`?

kmindspark, 2016-11-22:
/Users/myname/anaconda/bin:/Users/kaushikshivakumar/anaconda/bin:/usr/bin:/bin:/usr/sbin:/sbin:/usr/local/bin:/usr/local/apache-maven-3.3.9/bin:/Applications/PROS_2.0/gcc-arm-none-eabi-4_7-2014q2/bin:/Applications/PROS_2.0/gcc-arm-none-eabi-4_7-2014q2/bin:/Applications/PROS_2.0/gcc-arm-none-eabi-4_7-2014q2/bin:/Applications/PROS_2.0/gcc-arm-none-eabi-4_7-2014q2/bin:/Applications/PROS_2.0/gcc-arm-none-eabi-4_7-2014q2/bin:/Applications/PROS_2.0/gcc-arm-none-eabi-4_7-2014q2/bin:/Applications/PROS_2.0/gcc-arm-none-eabi-4_7-2014q2/bin

jwalk511, 2016-11-23:
This is my fault for not having the installer fail properly when Python isn't installed. Try downloading the newest version of Python 3 from https://www.python.org/downloads/ (It should Auto-detect your operating system). Then open Terminal and run `pip3 install pros-cli`. That should get you everything you nee; no need to rerun the installer.

kmindspark, 2016-11-23:
Thanks, this worked. However, after I created a new PROS project, each time I click on a ".c" file, it gives me an ENOENT error. Is there a way to fix this?

kmindspark, 2016-11-23:
Here are the full details:
Uncaught Error: ENOENT: no such file or directory, stat 'arm-none-eabi-g++'
fs.js:992
Hide Stack Trace
Error: ENOENT: no such file or directory, stat 'arm-none-eabi-g++'
    at Error (native)
    at fs.statSync (fs.js:992:18)
    at Object.fs.statSync (ELECTRON_ASAR.js:275:16)
    at /Users/kaushikshivakumar/.atom/packages/pros/lib/lint.coffee:53:38
    at /Users/kaushikshivakumar/.atom/packages/pros/node_modules/command-exists/lib/command-exists.js:12:13
    at ChildProcess.exithandler (child_process.js:225:5)
    at emitTwo (events.js:106:13)
    at ChildProcess.emit (events.js:191:7)
    at maybeClose (internal/child_process.js:877:16)
    at Socket.<anonymous> (internal/child_process.js:334:11)
    at emitOne (events.js:96:13)
    at Socket.emit (events.js:188:7)
    at Pipe._handle.close [as _onclose] (net.js:493:12)

jwalk511, 2016-11-23:
Just seeing that this involves lint.coffee, I think this is probably solved by downloading Xcode from the app store, and then running it at least once. 

Just as a sanity check, you should be able to open Terminal and run `arm-none-eabi-g++`, and get `arm-none-eabi-g++: fatal error: no input files
compilation terminated.` in response.

csmith105, 2016-12-05:
I;m getting this too, macOS Sierra.

TypeError: Cannot read property '1' of null
    at /Users/Cody/.atom/packages/pros/lib/views/welcome/welcome-view.coffee:75:98
    at ChildProcess.<anonymous> (/Users/Cody/.atom/packages/pros/lib/terminal-utilities.coffee:29:7)
    at emitTwo (events.js:100:13)
    at ChildProcess.emit (events.js:185:7)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:204:12)

Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.5/bin/pros", line 7, in <module>
    from proscli.main import main
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/proscli/__init__.py", line 5, in <module>
    from proscli.conductor import conductor_cli
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/proscli/conductor.py", line 77, in <module>
    @click.option('--registrar', metavar='REGISTRAR', prompt=True, type=click.Choice(available_providers()),
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/proscli/conductor.py", line 71, in available_providers
    return utils.get_all_provider_types().keys()
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/prosconductor/providers/utils.py", line 18, in get_all_provider_types
    spec.loader.load_module()
  File "<frozen importlib._bootstrap_external>", line 385, in _check_name_wrapper
  File "<frozen importlib._bootstrap_external>", line 806, in load_module
  File "<frozen importlib._bootstrap_external>", line 665, in load_module
  File "<frozen importlib._bootstrap>", line 265, in _load_module_shim
  File "<frozen importlib._bootstrap>", line 626, in _exec
  File "<frozen importlib._bootstrap_external>", line 658, in exec_module
  File "<frozen importlib._bootstrap_external>", line 868, in get_code
  File "<frozen importlib._bootstrap_external>", line 815, in get_data
FileNotFoundError: [Errno 2] No such file or directory: '/Applications/PROS_2.0/PROS_CLI/githubreleases.pyc'

HotelCalifornia, 2016-12-05:
Looks like your strace is keeping it 💯 :)

Just to confirm, have you tried the solution @jwalk511 posted above?

csmith105, 2016-12-05:
Yes I have both run XCode and I already had Python 3.5 installed. I ran the Python installer again just to be sure.

I also cannot create projects.

csmith105, 2016-12-05:
Also everytime I run Atom, I get this...

[image]

It seems to be trying to update but never succeeds.

edjubuh, 2016-12-05:
The CLI stack trace looks sort of similar to the issues we were having in purduesigbots/pros-cli#15. Can you run `pip3 install --upgrade pros-cli`, then `rm -rf ~/Library/Application Support/PROS` (@jwalk511 might need to confirm the location that this is where the PROS CLI local config is, although if it's not there it should be in ~/.pros)

csmith105, 2016-12-05:
```
Last login: Mon Dec  5 10:18:31 on ttys001
Codys-Mac:~ Cody$ pip3 install --upgrade pros-cli
Requirement already up-to-date: pros-cli in /Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages
Requirement already up-to-date: tabulate in /Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages (from pros-cli)
Requirement already up-to-date: semantic-version in /Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages (from pros-cli)
Requirement already up-to-date: jsonpickle in /Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages (from pros-cli)
Requirement already up-to-date: click in /Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages (from pros-cli)
Requirement already up-to-date: pyserial in /Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages (from pros-cli)
Requirement already up-to-date: cachetools in /Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages (from pros-cli)
Requirement already up-to-date: requests in /Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages (from pros-cli)
You are using pip version 8.1.1, however version 9.0.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
Codys-Mac:~ Cody$ rm -rf ~/Library/Application Support/PROS
Codys-Mac:~ Cody$ 
```

No change.

jwalk511, 2016-12-06:
`~/Library/Application\ Support/Pros is the correct path. I don't know how Bash will respond if you don't escape the space in 'Application Support', but it probably doesn't do what we want. Try `rm -rf ~/Library/Application Support/PROS`, and see how that works.

Since you managed to get one of the early installers, if you run `which pros` you may not see `/Library/Frameworks/Python.framework/Versions/3.5/bin/pros`
If you see a path with `Applications`, then modify `~/.bash_profile` and erase the line that contains `PROS/PROS_CLI`
Save and restart Terminal

If `which pros` still doesn't give you the right path, try `pip3 uninstall pros-cli` and then `pip3 install --upgrade pros-cli`

As for Atom, the update issue is probably it fussing about living in an admin-owned space (/Applications), and I might have to start installing it in `~/Applications` instead. Try moving Atom to your desktop or some other user-owned space, and see if you still have the issue. If you do, run `sudo /Applications/PROS_2.0/scripts/uninstallAtom.command` and try again. Once you do, you'll have to re download the PROS plugin from Atom's package manager. For future reminder, `apm install pros` from the command line, or just search `pros` in Atom's package installer.

HotelCalifornia, 2016-12-06:
As the original issue in this thread was solved above, I am going to create new issues for each separate problem with the hope that we can keep relevant discourse in relevant places.

To recap, the original issue was a stacktrace containing the following error:

    TypeError: Cannot read property '1' of null
        at /Users/kaushikshivakumar/.atom/packages/pros/lib/views/welcome/welcome-view.coffee:75:98
        at ChildProcess.<anonymous> (/Users/kaushikshivakumar/.atom/packages/pros/lib/terminal-utilities.coffee:29:7)
        at emitTwo (events.js:106:13)
        at ChildProcess.emit (events.js:191:7)
        at Process.ChildProcess._handle.onexit (internal/child_process.js:215:12)

which was solved in the following post:

> This is my fault for not having the installer fail properly when Python isn't installed. Try downloading the newest version of Python 3 from https://www.python.org/downloads/ (It should Auto-detect your operating system). Then open Terminal and run pip3 install --upgrade pros-cli. That should get you everything you nee; no need to rerun the installer. [per @jwalk511]

The next issue involving the linter throwing an ENOENT error can now be found in #19

closed 2016-12-06 by HotelCalifornia