purduesigbots/pros-atom #18: Uncaught TypeError: Cannot read property '1' of null

TimZhang9594, 2016-11-21:
[Enter steps to reproduce below:]

1. ...
2. ...

**Atom Version**: 1.12.4
**Electron Version**: 1.3.9
**System**: Ubuntu 16.04.1
**Thrown From**: [pros](https://github.com/purduesigbots/pros-atom) package, v0.5.4

### Stack Trace

Uncaught TypeError: Cannot read property '1' of null

```
At /home/timzhang/.atom/packages/pros/lib/views/welcome/welcome-view.coffee:75

TypeError: Cannot read property '1' of null
    at /home/timzhang/.atom/packages/pros/lib/views/welcome/welcome-view.coffee:75:98
    at ChildProcess.<anonymous> (/home/timzhang/.atom/packages/pros/lib/terminal-utilities.coffee:29:7)
    at emitTwo (events.js:106:13)
    at ChildProcess.emit (events.js:191:7)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:215:12)
```

### Commands

```
```

### Config

```json
{
  "core": {
    "telemetryConsent": "limited"
  },
  "pros": {
    "googleAnalytics": {
      "cid": "a4897648-482e-4581-90aa-2c20d7911e4e"
    }
  }
}
```

### Installed Packages

```coffee
# User
build, v0.67.0 (active)
busy, v0.7.0 (active)
file-icons, v1.7.25 (active)
linter, v1.11.18 (active)
platformio-ide-terminal, v2.2.2 (active)
pros, v0.5.4 (active)
tool-bar, v1.0.1 (active)
tool-bar-main, v0.0.10 (active)
atom-dark-syntax, v0.27.0 (inactive)
atom-dark-ui, v0.52.0 (inactive)
atom-light-syntax, v0.28.0 (inactive)
atom-light-ui, v0.45.0 (inactive)
base16-tomorrow-dark-theme, v1.3.0 (inactive)
base16-tomorrow-light-theme, v1.3.0 (inactive)
one-dark-ui, v1.6.2 (active)
one-light-ui, v1.6.2 (inactive)
one-dark-syntax, v1.5.0 (active)
one-light-syntax, v1.5.0 (inactive)
solarized-dark-syntax, v1.0.5 (inactive)
solarized-light-syntax, v1.0.5 (inactive)
about, v1.7.0 (active)
archive-view, v0.62.0 (active)
autocomplete-atom-api, v0.10.0 (active)
autocomplete-css, v0.13.1 (active)
autocomplete-html, v0.7.2 (active)
autocomplete-plus, v2.31.4 (active)
autocomplete-snippets, v1.11.0 (active)
autoflow, v0.27.0 (inactive)
autosave, v0.23.1 (active)
background-tips, v0.26.1 (active)
bookmarks, v0.42.0 (active)
bracket-matcher, v0.82.2 (active)
command-palette, v0.39.0 (inactive)
deprecation-cop, v0.54.1 (active)
dev-live-reload, v0.47.0 (active)
encoding-selector, v0.22.0 (active)
exception-reporting, v0.40.0 (active)
find-and-replace, v0.202.2 (inactive)
fuzzy-finder, v1.4.0 (active)
git-diff, v1.1.0 (active)
go-to-line, v0.31.0 (inactive)
grammar-selector, v0.48.2 (active)
image-view, v0.60.0 (active)
incompatible-packages, v0.26.1 (active)
keybinding-resolver, v0.35.0 (active)
line-ending-selector, v0.5.0 (active)
link, v0.31.2 (inactive)
markdown-preview, v0.158.8 (active)
metrics, v1.1.2 (active)
notifications, v0.65.1 (active)
open-on-github, v1.2.1 (inactive)
package-generator, v1.0.1 (inactive)
settings-view, v0.243.1 (active)
snippets, v1.0.4 (active)
spell-check, v0.68.4 (active)
status-bar, v1.4.1 (active)
styleguide, v0.47.2 (active)
symbols-view, v0.113.1 (inactive)
tabs, v0.103.0 (active)
timecop, v0.33.2 (active)
tree-view, v0.210.0 (active)
update-package-dependencies, v0.10.0 (active)
welcome, v0.35.1 (active)
whitespace, v0.35.0 (active)
wrap-guide, v0.38.2 (active)
language-c, v0.54.0 (active)
language-clojure, v0.22.1 (active)
language-coffee-script, v0.48.0 (active)
language-csharp, v0.13.0 (active)
language-css, v0.40.1 (active)
language-gfm, v0.88.0 (active)
language-git, v0.15.0 (active)
language-go, v0.43.0 (active)
language-html, v0.46.1 (active)
language-hyperlink, v0.16.1 (active)
language-java, v0.24.0 (active)
language-javascript, v0.122.0 (active)
language-json, v0.18.3 (active)
language-less, v0.29.6 (active)
language-make, v0.22.2 (active)
language-mustache, v0.13.0 (active)
language-objective-c, v0.15.1 (active)
language-perl, v0.37.0 (active)
language-php, v0.37.3 (active)
language-property-list, v0.8.0 (active)
language-python, v0.45.1 (active)
language-ruby, v0.70.2 (active)
language-ruby-on-rails, v0.25.1 (active)
language-sass, v0.57.0 (active)
language-shellscript, v0.23.0 (active)
language-source, v0.9.0 (active)
language-sql, v0.25.0 (active)
language-text, v0.7.1 (active)
language-todo, v0.29.1 (active)
language-toml, v0.18.1 (active)
language-xml, v0.34.12 (active)
language-yaml, v0.27.1 (active)

# Dev
No dev packages
```

HotelCalifornia, 2016-11-21:
See #17

TimZhang9594, 2016-11-21:
#17  is MacOs and I am using Ubuntu

HotelCalifornia, 2016-11-21:
Doesn't matter, the issue is the same

edjubuh, 2016-11-21:
@TimZhang9594, how did you install PROS for Atom?

TimZhang9594, 2016-11-21:
@edjubuh I copied the sudo codes from the downloads (linux.sh something like that)

jabibi100, 2016-11-21:
@TimZhang9594 is there a particular reason why you did not just run the script we provided for Ubuntu?

To execute the script simply type `./pros-install_ubuntu.sh` in the directory that you downloaded the script and enter your `sudo` password when prompted.

TimZhang9594, 2016-11-21:
@jabibi100 Do you mind telling me how exactly to do it, because I am pretty new to ubuntu

HotelCalifornia, 2016-11-21:
@TimZhang9594
Exactly as he said:
"To execute the script simply type ./pros-install_ubuntu.sh in the directory that you downloaded the script and enter your sudo password when prompted."

HotelCalifornia, 2016-11-21:
Reopening this issue in case it really is different than the MacOS one

jabibi100, 2016-11-21:
@TimZhang9594 I'll expand my instructions now for more details. Open a terminal and run the following commands:

`cd ~`
`wget https://github.com/purduesigbots/pros/releases/download/2.11.0/pros-install_ubuntu.sh`
`./pros-install_ubuntu.sh`

The first command changes the current directory to your home directory. The second command downloads the install script from github. Finally, the third command runs the script. As previous enter your sudo password when prompted after executing the script.

edjubuh, 2017-01-26:
I'm going to mark this issue as closed since @TimZhang9594 hasn't responded otherwise.

closed 2017-01-26 by edjubuh