purduesigbots/pros-atom #20: Unable to Update Atom (macOS

jwalk511, 2016-12-09:
#### Expected Behavior:
Clicking the `Restart and Install Update` button in the `Atom` menu should restart Atom and launch the updated version
#### Actual Behavior:
Clicking the button restarts Atom and launches the old version.
#### Steps to reproduce:
Have Atom installed in any directory where it does not have write permission.
#### System information:
Operating System:  (Windows, OS X, Ubuntu)
Found on macOS. May apply to Ubuntu, depending on where it is installed.
Atom Version: (Atom Version)
Any
Plugin Version: (Plugin Version)
N/A
#### Additional Information
Default location for Atom on macOS through PROS installer is in `/Applications/PROS_2.0`. By default, Atom does not have permission to write in `/Applications` or its subdirectories. Atom either needs to be installed somewhere else, or the permissions of the directory need to be changed.
#### Screenshots

jwalk511, 2016-12-10:
See https://github.com/atom/atom/issues/5692

This is a known issue. It looks like the best way is going to be to install Atom somewhere else and drop and Alias in /Applications/PROS_2.0

HotelCalifornia, 2017-10-05:
Should look into this again after we release the current round of installers.

HotelCalifornia, 2017-10-05:
Actually, I think this will be a moot point WRT `PROS Editor` as I believe we're going to be pinning Atom at a specific version, and updating when we're ready. cc @edjubuh correct me if I'm wrong on that

closed 2017-11-26 by edjubuh