purduesigbots/pros-atom #23: cli.uploadInTerminal and cli.serialInTerminal are not functions

sigbot, 2017-01-19:
Confirmed on two computers so far-- the lab computer and Brian's laptop.
To repro(??) try using the pros commands from within Atom.

Further, there may be a separate issue with the CLI, as the lab computer throws a fit whenever I try to run a serial-related command, though this has not been seen on other computers.

Neither restarting nor running the repair utility from the installer seem to have worked.

**Atom Version**: 1.13.0
**Electron Version**: 1.3.13
**System**: Microsoft Windows 10 Education
**Thrown From**: [pros](https://github.com/purduesigbots/pros-atom) package, v0.5.4

### Stack Trace

Uncaught TypeError: cli.uploadInTerminal is not a function

```
At C:\Users\BLRS\Desktop\pros-atom\lib\main.coffee:141

TypeError: cli.uploadInTerminal is not a function
    at Object.uploadProject (file:///C:/Users/BLRS/Desktop/pros-atom/lib/main.coffee:141:11)
    at atom-workspace.PROS:Upload-Project (file:///C:/Users/BLRS/Desktop/pros-atom/lib/main.coffee:67:70)
    at CommandRegistry.module.exports.CommandRegistry.handleCommandEvent (C:\Users\BLRS\AppData\Local\atom\app-1.13.0\resources\app.asar\src\command-registry.js:259:29)
    at CommandRegistry.handleCommandEvent (C:\Users\BLRS\AppData\Local\atom\app-1.13.0\resources\app.asar\src\command-registry.js:3:59)
    at CommandRegistry.module.exports.CommandRegistry.dispatch (C:\Users\BLRS\AppData\Local\atom\app-1.13.0\resources\app.asar\src\command-registry.js:160:19)
    at executeCallback (C:/Users/BLRS/.atom/packages/tool-bar/lib/tool-bar-button-view.js:105:19)
    at ToolBarButtonView._onClick (C:/Users/BLRS/.atom/packages/tool-bar/lib/tool-bar-button-view.js:69:7)
```

edjubuh, 2017-01-19:
Tag on #21 . Likely an issue where these functions haven't been migrated to the new `proscli` module

edjubuh, 2017-01-26:
Fixed by 39c8c8b

closed 2017-01-26 by edjubuh