purduesigbots/pros-atom #26: PROS on my Mac is skullfucked

csmith105, 2017-02-03:
Highly technical title I know, let's explain the situation.

I have myself a late 2013 15" MBP (the retina one) (OS X 10.12.2) that's never been formatted and has been through 2.5 years of a CS degree, involving setting up and writing code for everything under the sun, Java, C, C++, C#, Python 2.7, Python 3.1, Python 3.14, Python 3.1415, Ada, PHP, etc...

All those blindly enter the following commands as root tend to eat at the core of machines like this.

#### Expected Behavior:

Ability to actually make a PROS project.

#### Actual Behavior:

Launching Atom produces an error:
[http://polynomic3d.com/user/smith/Screen%20Shot%202017-02-03%20at%203.37.00%20PM.png](url)

Trying to create a new project produces an error:
[http://polynomic3d.com/user/smith/Screen%20Shot%202017-02-03%20at%203.37.09%20PM.png](url)

Continuing to try to create a new project produces an error:
[http://polynomic3d.com/user/smith/Screen%20Shot%202017-02-03%20at%203.37.33%20PM.png](url)

Notice that it cannot find a version of the Kernel, it's never been able to. I'll be happy to let you guys poke around my system if need be.

**Atom Version**: 1.13.1
**Electron Version**: 1.3.13
**System**: Mac OS X 10.12.2
**Thrown From**: [pros](https://github.com/purduesigbots/pros-atom) package, v0.6.1

### Stack Trace

Uncaught SyntaxError: Unexpected token L in JSON at position 0

```
At file:///Applications/Atom.app/Contents/Resources/app.asar/static/index.html#%7B%22initialPaths%22%3A%5B%5D%2C%22locationsToOpen%22%3A%5B%7B%7D%5D%2C%22windowInitializationScript%22%3A%22%2FApplications%2FAtom.app%2FContents%2FResources%2Fapp.asar%2Fsrc%2Finitialize-application-window.js%22%2C%22resourcePath%22%3A%22%2FApplications%2FAtom.app%2FContents%2FResources%2Fapp.asar%22%2C%22devMode%22%3Afalse%2C%22safeMode%22%3Afalse%2C%22profileStartup%22%3Afalse%2C%22clearWindowState%22%3Afalse%2C%22env%22%3A%7B%22TMPDIR%22%3A%22%2Fvar%2Ffolders%2F49%2Ftnt6_xx918vb5jq4dqm0f4s80000gn%2FT%2F%22%2C%22__CF_USER_TEXT_ENCODING%22%3A%220x1F5%3A0x0%3A0x0%22%2C%22SHELL%22%3A%22%2Fbin%2Fbash%22%2C%22HOME%22%3A%22%2FUsers%2FCody%22%2C%22Apple_PubSub_Socket_Render%22%3A%22%2Fprivate%2Ftmp%2Fcom.apple.launchd.BfZz4YjP0X%2FRender%22%2C%22SSH_AUTH_SOCK%22%3A%22%2Fprivate%2Ftmp%2Fcom.apple.launchd.Uo1SMROUmk%2FListeners%22%2C%22PATH%22%3A%22%2Fusr%2Fbin%3A%2Fbin%3A%2Fusr%2Fsbin%3A%2Fsbin%22%2C%22LOGNAME%22%3A%22Cody%22%2C%22XPC_SERVICE_NAME%22%3A%22com.github.atom.13912%22%2C%22USER%22%3A%22Cody%22%2C%22XPC_FLAGS%22%3A%220x0%22%2C%22GOOGLE_API_KEY%22%3A%22AIzaSyAQfxPJiounkhOjODEO5ZieffeBv6yft2Q%22%2C%22ATOM_HOME%22%3A%22%2FUsers%2FCody%2F.atom%22%7D%2C%22appVersion%22%3A%221.13.1%22%2C%22atomHome%22%3A%22%2FUsers%2FCody%2F.atom%22%2C%22shellLoadTime%22%3A544%7D:1

SyntaxError: Unexpected token L in JSON at position 0
    at Object.parse (native)
    at HTMLButtonElement.<anonymous> (/Users/Cody/.atom/packages/pros/lib/views/new-project.coffee:46:27)
    at HTMLButtonElement.dispatch (/Users/Cody/.atom/packages/pros/node_modules/jquery/dist/jquery.js:4435:9)
    at HTMLButtonElement.elemData.handle (/Users/Cody/.atom/packages/pros/node_modules/jquery/dist/jquery.js:4121:28)
```

### Commands

```
     -0:33.2.0 tree-view:show (atom-workspace.workspace.scrollbars-visible-when-scrolling.theme-one-dark-syntax.theme-one-dark-ui)
     -0:23.9.0 PROS:New-Project (atom-pane.pane.active)
     -0:12.1.0 core:cancel (atom-pane.pane.active)
```

### Config

```json
{
  "core": {
    "telemetryConsent": "limited"
  },
  "pros": {
    "googleAnalytics": {
      "cid": "2f903c63-b1b6-422e-a68f-11c778adbb38"
    }
  }
}
```

### Installed Packages

```coffee
# User
atom-beautify, v0.29.17 (inactive)
build, v0.67.0 (active)
busy, v0.7.0 (active)
file-icons, v2.0.13 (active)
linter, v1.11.18 (active)
platformio-ide-terminal, v2.2.2 (active)
pros, v0.6.1 (active)
tool-bar, v1.0.1 (active)
tool-bar-main, v0.0.10 (active)
atom-dark-syntax, v0.28.0 (inactive)
atom-dark-ui, v0.53.0 (inactive)
atom-light-syntax, v0.29.0 (inactive)
atom-light-ui, v0.46.0 (inactive)
base16-tomorrow-dark-theme, v1.4.0 (inactive)
base16-tomorrow-light-theme, v1.4.0 (inactive)
one-dark-ui, v1.8.2 (active)
one-light-ui, v1.8.2 (inactive)
one-dark-syntax, v1.6.0 (active)
one-light-syntax, v1.6.0 (inactive)
solarized-dark-syntax, v1.1.1 (inactive)
solarized-light-syntax, v1.1.1 (inactive)
about, v1.7.2 (active)
archive-view, v0.62.0 (active)
autocomplete-atom-api, v0.10.0 (active)
autocomplete-css, v0.14.1 (active)
autocomplete-html, v0.7.2 (active)
autocomplete-plus, v2.33.1 (active)
autocomplete-snippets, v1.11.0 (active)
autoflow, v0.29.0 (inactive)
autosave, v0.23.2 (active)
background-tips, v0.26.1 (active)
bookmarks, v0.43.2 (active)
bracket-matcher, v0.85.1 (active)
command-palette, v0.39.1 (inactive)
deprecation-cop, v0.55.1 (active)
dev-live-reload, v0.47.0 (active)
encoding-selector, v0.22.0 (active)
exception-reporting, v0.40.0 (active)
find-and-replace, v0.204.5 (inactive)
fuzzy-finder, v1.4.0 (active)
git-diff, v1.2.0 (active)
go-to-line, v0.31.2 (inactive)
grammar-selector, v0.48.2 (active)
image-view, v0.60.0 (active)
incompatible-packages, v0.26.1 (active)
keybinding-resolver, v0.35.0 (active)
line-ending-selector, v0.5.1 (active)
link, v0.31.2 (inactive)
markdown-preview, v0.159.3 (active)
metrics, v1.1.3 (active)
notifications, v0.65.1 (active)
open-on-github, v1.2.1 (inactive)
package-generator, v1.0.2 (inactive)
settings-view, v0.244.0 (active)
snippets, v1.0.4 (active)
spell-check, v0.68.5 (active)
status-bar, v1.6.0 (active)
styleguide, v0.48.0 (active)
symbols-view, v0.113.1 (inactive)
tabs, v0.103.1 (active)
timecop, v0.33.2 (active)
tree-view, v0.211.1 (active)
update-package-dependencies, v0.10.0 (active)
welcome, v0.35.1 (active)
whitespace, v0.35.0 (active)
wrap-guide, v0.39.0 (active)
language-c, v0.54.0 (active)
language-clojure, v0.22.1 (active)
language-coffee-script, v0.48.1 (active)
language-csharp, v0.13.0 (active)
language-css, v0.40.1 (active)
language-gfm, v0.88.0 (active)
language-git, v0.15.0 (active)
language-go, v0.43.0 (active)
language-html, v0.47.1 (active)
language-hyperlink, v0.16.1 (active)
language-java, v0.24.0 (active)
language-javascript, v0.122.0 (active)
language-json, v0.18.3 (active)
language-less, v0.29.6 (active)
language-make, v0.22.2 (active)
language-mustache, v0.13.0 (active)
language-objective-c, v0.15.1 (active)
language-perl, v0.37.0 (active)
language-php, v0.37.3 (active)
language-property-list, v0.8.0 (active)
language-python, v0.45.1 (active)
language-ruby, v0.70.2 (active)
language-ruby-on-rails, v0.25.1 (active)
language-sass, v0.57.0 (active)
language-shellscript, v0.23.0 (active)
language-source, v0.9.0 (active)
language-sql, v0.25.0 (active)
language-text, v0.7.1 (active)
language-todo, v0.29.1 (active)
language-toml, v0.18.1 (active)
language-xml, v0.34.12 (active)
language-yaml, v0.27.1 (active)

# Dev
No dev packages
```

#### Steps to reproduce:

I've installed PROS, installed Atom seperately, updated Atom and made a symbolic link in the PROS dir.

#### System information:
Operating System:  (Windows, OS X, Ubuntu)

OS X 10.12.2

Atom Version: (Atom Version) / Plugin Version: (Plugin Version)

**Atom Version**: 1.13.1
**Electron Version**: 1.3.13
**System**: Mac OS X 10.12.2
**Thrown From**: [pros](https://github.com/purduesigbots/pros-atom) package, v0.6.1

HotelCalifornia, 2017-02-03:
I couldn't get those image URLs to open properly, so I'll drop them here:

>> Launching Atom produces an error:
>> [image]
>
>>Trying to create a new project produces an error:
>> [image]
>
>> Continuing to try to create a new project produces an error:
>> [image]

HotelCalifornia, 2017-02-03:
Can you try/have you already tried running the the relevant commands in a terminal?

e.g. 
`pros conduct new project`

csmith105, 2017-02-03:
http://polynomic3d.com/user/smith/Screen%20Shot%202017-02-03%20at%204.26.10%20PM.png

HotelCalifornia, 2017-02-03:
Our resident Mac dev has asked me to ask you to run the following command to attempt to upgrade the CLI:
```
pip3 install --upgrade pros-cli
```

csmith105, 2017-02-03:
`Last login: Thu Feb  2 14:53:14 on ttys001
Restored session: Fri Feb  3 15:35:38 MST 2017
Codys-Mac:~ Cody$ pip3 install --upgrade pros-cli
Requirement already up-to-date: pros-cli in /Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages
Requirement already up-to-date: semantic-version in /Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages (from pros-cli)
Requirement already up-to-date: cachetools in /Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages (from pros-cli)
Requirement already up-to-date: pyserial in /Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages (from pros-cli)
Requirement already up-to-date: click in /Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages (from pros-cli)
Requirement already up-to-date: requests in /Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages (from pros-cli)
Requirement already up-to-date: jsonpickle in /Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages (from pros-cli)
Requirement already up-to-date: tabulate in /Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages (from pros-cli)
Codys-Mac:~ Cody$ 
`

csmith105, 2017-02-03:
It appears my entire PROS_CLI folder is missing...

`Codys-Mac:PROS_2.0 Cody$ sudo ls -l
total 8
-rw-r--r--@ 1 Cody  staff  728 Feb  1 16:33 Atom
drwxr-xr-x  6 root  admin  204 Oct 19 20:29 gcc-arm-none-eabi-4_7-2014q2
drwxr-xr-x  4 root  admin  136 Oct 19 20:29 scripts
Codys-Mac:PROS_2.0 Cody$ `

jwalk511, 2017-02-03:
Go into `~/Library/Application\ Support/PROS/` and delete `cli.pros`. This should be what is pointing to the invalid path for `githubreleases.pyc`

jwalk511, 2017-02-03:
You are correct at the bottom of your first comment, this is something that was left over from an older installation of PROS that the installer does not overwrite.

csmith105, 2017-02-03:
That seemed to have worked! I can now create projects and I'm seeing Kernel versions.

One small addendum is that I'm getting this:

http://polynomic3d.com/user/smith/Screen%20Shot%202017-02-03%20at%204.44.16%20PM.png

Even though `pip3 install --upgrade pros-cli` shows everything up to date. Clicking the install button isn't doing anything.

edjubuh, 2017-02-03:
We messed up on this particular release with versioning - we're moving to a fully automated Jenkins build system, not that that's any excuse. I'm working on getting everything back up and running.

csmith105, 2017-02-03:
No biggie, thanks so much guys! I'm going to close this now.