purduesigbots/pros-atom #27: Uncaught Error: ENOENT: no such file or directory, stat 'arm-none-eabi-g  '

Jaynard2, 2017-02-06:
[Enter steps to reproduce below:]

1. ...
2. ...

**Atom Version**: 1.13.1
**Electron Version**: 1.3.13
**System**: Mac OS X 10.12.3
**Thrown From**: [pros](https://github.com/purduesigbots/pros-atom) package, v0.6.1

### Stack Trace

Uncaught Error: ENOENT: no such file or directory, stat 'arm-none-eabi-g  '

```
At fs.js:992

Error: ENOENT: no such file or directory, stat 'arm-none-eabi-g  '
    at Error (native)
    at fs.statSync (fs.js:992:18)
    at Object.fs.statSync (ELECTRON_ASAR.js:275:16)
    at /Users/Joshua/.atom/packages/pros/lib/lint.coffee:50:38
    at /Users/Joshua/.atom/packages/pros/node_modules/command-exists/lib/command-exists.js:12:13
    at ChildProcess.exithandler (child_process.js:225:5)
    at emitTwo (events.js:106:13)
    at ChildProcess.emit (events.js:191:7)
    at maybeClose (internal/child_process.js:877:16)
    at Socket.<anonymous> (internal/child_process.js:334:11)
    at emitOne (events.js:96:13)
    at Socket.emit (events.js:188:7)
    at Pipe._handle.close [as _onclose] (net.js:493:12)
```

### Commands

```
     -1:18.5.0 tree-view:show (atom-workspace.workspace.scrollbars-visible-when-scrolling.theme-atom-dark-syntax.theme-one-dark-ui)
```

### Config

```json
{
  "core": {
    "openEmptyEditorOnStart": false,
    "projectHome": "/Users/Joshua/vexRobotics/PROS",
    "telemetryConsent": "limited",
    "themes": [
      "one-dark-ui",
      "atom-dark-syntax"
    ]
  },
  "pros": {
    "googleAnalytics": {
      "cid": "aa42216b-0b2b-40a5-8fa7-a9bf1e56fe3a"
    }
  }
}
```

### Installed Packages

```coffee
# User
atom-beautify, v0.29.17 (inactive)
build, v0.67.0 (active)
busy, v0.7.0 (active)
file-icons, v2.0.14 (active)
linter, v1.11.21 (active)
platformio-ide-terminal, v2.2.2 (active)
project-manager, v3.3.3 (active)
pros, v0.6.1 (active)
tool-bar, v1.0.1 (active)
tool-bar-main, v0.0.10 (active)
atom-dark-syntax, v0.28.0 (active)
atom-dark-ui, v0.53.0 (inactive)
atom-light-syntax, v0.29.0 (inactive)
atom-light-ui, v0.46.0 (inactive)
base16-tomorrow-dark-theme, v1.4.0 (inactive)
base16-tomorrow-light-theme, v1.4.0 (inactive)
one-dark-ui, v1.8.2 (active)
one-light-ui, v1.8.2 (inactive)
one-dark-syntax, v1.6.0 (inactive)
one-light-syntax, v1.6.0 (inactive)
solarized-dark-syntax, v1.1.1 (inactive)
solarized-light-syntax, v1.1.1 (inactive)
about, v1.7.2 (active)
archive-view, v0.62.0 (active)
autocomplete-atom-api, v0.10.0 (active)
autocomplete-css, v0.14.1 (active)
autocomplete-html, v0.7.2 (active)
autocomplete-plus, v2.33.1 (active)
autocomplete-snippets, v1.11.0 (active)
autoflow, v0.29.0 (inactive)
autosave, v0.23.2 (active)
background-tips, v0.26.1 (active)
bookmarks, v0.43.2 (active)
bracket-matcher, v0.85.1 (active)
command-palette, v0.39.1 (inactive)
deprecation-cop, v0.55.1 (active)
dev-live-reload, v0.47.0 (active)
encoding-selector, v0.22.0 (active)
exception-reporting, v0.40.0 (active)
find-and-replace, v0.204.5 (inactive)
fuzzy-finder, v1.4.0 (active)
git-diff, v1.2.0 (active)
go-to-line, v0.31.2 (inactive)
grammar-selector, v0.48.2 (active)
image-view, v0.60.0 (active)
incompatible-packages, v0.26.1 (active)
keybinding-resolver, v0.35.0 (active)
line-ending-selector, v0.5.1 (active)
link, v0.31.2 (inactive)
markdown-preview, v0.159.3 (active)
metrics, v1.1.3 (active)
notifications, v0.65.1 (active)
open-on-github, v1.2.1 (inactive)
package-generator, v1.0.2 (inactive)
settings-view, v0.244.0 (active)
snippets, v1.0.4 (active)
spell-check, v0.68.5 (active)
status-bar, v1.6.0 (active)
styleguide, v0.48.0 (active)
symbols-view, v0.113.1 (inactive)
tabs, v0.103.1 (active)
timecop, v0.33.2 (active)
tree-view, v0.211.1 (active)
update-package-dependencies, v0.10.0 (active)
welcome, v0.35.1 (active)
whitespace, v0.35.0 (active)
wrap-guide, v0.39.0 (active)
language-c, v0.54.0 (active)
language-clojure, v0.22.1 (active)
language-coffee-script, v0.48.1 (active)
language-csharp, v0.13.0 (active)
language-css, v0.40.1 (active)
language-gfm, v0.88.0 (active)
language-git, v0.15.0 (active)
language-go, v0.43.0 (active)
language-html, v0.47.1 (active)
language-hyperlink, v0.16.1 (active)
language-java, v0.24.0 (active)
language-javascript, v0.122.0 (active)
language-json, v0.18.3 (active)
language-less, v0.29.6 (active)
language-make, v0.22.2 (active)
language-mustache, v0.13.0 (active)
language-objective-c, v0.15.1 (active)
language-perl, v0.37.0 (active)
language-php, v0.37.3 (active)
language-property-list, v0.8.0 (active)
language-python, v0.45.1 (active)
language-ruby, v0.70.2 (active)
language-ruby-on-rails, v0.25.1 (active)
language-sass, v0.57.0 (active)
language-shellscript, v0.23.0 (active)
language-source, v0.9.0 (active)
language-sql, v0.25.0 (active)
language-text, v0.7.1 (active)
language-todo, v0.29.1 (active)
language-toml, v0.18.1 (active)
language-xml, v0.34.12 (active)
language-yaml, v0.27.1 (active)

# Dev
No dev packages
```

HotelCalifornia, 2017-02-13:
#30 seemed to have been fixed by updating to the latest version (0.6.3). Does this work for you too?

Jaynard2, 2017-02-15:
No, it hasn't.

jwalk511, 2017-02-15:
Please check whether the Arm toolchain is on your PATH or not

1. Open Terminal
2. Type `arm-none-eabi-gcc`
If you get ```arm-none-eabi-gcc: fatal error: no input files
compilation terminated.``` in response, then it is on your PATH. Let us know, and ignore the rest of this post.
If you get `-bash: arm-none-eabi-gcc: command not found`, then it is not in PATH.
3. If the command was not found, Type `$PATH` and paste the results in your response.

You can try to add it to PATH yourself by downloading the Mac Tarball from https://launchpad.net/gcc-arm-embedded/+download and then adding the download location to PATH by following the directions at https://coolestguidesontheplanet.com/add-shell-path-osx/
Note: You should move the toolchain out of your Downloads directory before adding it to path.

HotelCalifornia, 2017-02-20:
@Jaynard2 did any of the above fix your issue, or is it still happening?

Jaynard2, 2017-02-20:
Sorry for the delayed response; I had a few things come up. No this hasn't fixed my issue. I downloaded the Mac Tarball and set the PATH to the downloaded file, the file that it is in, and I tried extracting it and setting the path to there. Here is my PATH:
/PROS_2.0/gcc-arm-none-eabi-5_4-2016q3-20160926-mac.tar.bz2 :/Library/Frameworks/Python.framework/Versions/3.6/bin:/Library/Frameworks/Python.framework/Versions/3.6/bin:/Library/Frameworks/Python.framework/Versions/3.6/bin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin:/opt/X11/bin:/Applications/PROS_2.0/gcc-arm-none-eabi-4_7-2014q2/bin:/Applications/PROS_2.0/gcc-arm-none-eabi-4_7-2014q2/bin:/Applications/PROS_2.0/gcc-arm-none-eabi-4_7-2014q2/bin

jwalk511, 2017-02-23:
Is `/PROS_2.0/gcc-arm-none-eabi-5_4-2016q3-20160926-mac.tar.bz2 ` a folder in your root directory? And why does it have a space at the end? Try removing it from your PATH, or at minimum deleting the space.

Jaynard2, 2017-02-24:
Yes, it is. I removed the space, and the command is still not found.

HotelCalifornia, 2017-04-02:
@Jaynard2 are you still having this issue?

HotelCalifornia, 2017-05-02:
Closing due to lack of activity

closed 2017-05-02 by HotelCalifornia