purduesigbots/pros-atom #29: Uncaught TypeError: cli.serialInTerminal is not a function

dabinaryboy, 2017-02-08:
[Enter steps to reproduce:]

1. ...
2. ...

**Atom**: 1.14.0 ia32
**Electron**: 1.3.13
**OS**: Microsoft Windows 10 Pro
**Thrown From**: [pros](https://github.com/purduesigbots/pros-atom) package 0.6.1

### Stack Trace

Uncaught TypeError: cli.serialInTerminal is not a function

```
At /C:/Users/Admin/.atom/packages/pros/lib/main.coffee:249

TypeError: cli.serialInTerminal is not a function
    at Object.toggleTerminal (/packages/pros/lib/main.coffee:146:26)
    at /packages/pros/lib/main.coffee:67:71)
    at CommandRegistry.module.exports.CommandRegistry.handleCommandEvent (/app.asar/src/command-registry.js:259:29)
    at CommandRegistry.handleCommandEvent (/app.asar/src/command-registry.js:3:59)
    at CommandRegistry.module.exports.CommandRegistry.dispatch (/app.asar/src/command-registry.js:160:19)
    at executeCallback (/packages/tool-bar/lib/tool-bar-button-view.js:105:19)
    at ToolBarButtonView._onClick (/packages/tool-bar/lib/tool-bar-button-view.js:69:7)
```

### Commands

```
     -0:24.3.0 tree-view:show (atom-workspace.workspace.scrollbars-visible-always.theme-one-dark-syntax.theme-one-dark-ui)
     -0:08.9.0 PROS:Toggle-Terminal (input.hidden-input)
```

### Non-Core Packages

```
build 0.67.0 
busy 0.7.0 
file-icons 2.0.15 
language-vex 0.1.3 
linter 1.11.21 
platformio-ide-terminal 2.2.2 
pros 0.6.1 
tool-bar 1.0.1 
tool-bar-main 0.0.10 
```

HotelCalifornia, 2017-02-09:
This issue is fixed in the latest (0.6.3) version of PROS for Atom. Can you try downloading the latest installer from [here](https://github.com/purduesigbots/pros/releases/download/2.11.0/pros-win.exe) and see if that fixes your problems?

Superwolf12354, 2017-02-16:
Uh.. Hi, I'm having the same problem, Except i'm on version 0.6.3. I even tried updating it manually, but it gives the same error.

commit 6cfaadce (2017-02-17): fix #29
commit 8882dd72 (2017-02-17): Merge pull request #34 from purduesigbots/more-cli-migration-bugfixes

fix #29
closed 2017-02-17 by HotelCalifornia