purduesigbots/pros-atom #37: Uncaught SyntaxError: Unexpected token L in JSON at position 0

Reichenbachian, 2017-02-27:
I've just been trying to get PROS to work with Atom on the mac, and it really doesn't want to. I finally got the thing in and loaded, but now when I try to download the new kernels, this error happens. All I do is click "Create new PROS Project".

**Atom**: 1.14.3 x64
**Electron**: 1.3.13
**OS**: Mac OS X 10.12.3
**Thrown From**: [pros](https://github.com/purduesigbots/pros-atom) package 0.6.5

### Stack Trace

Uncaught SyntaxError: Unexpected token L in JSON at position 0

```
At file:///Applications/Atom.app/Contents/Resources/app.asar/static/index.html#%7B%22initialPaths%22%3A%5B%5D%2C%22locationsToOpen%22%3A%5B%7B%7D%5D%2C%22windowInitializationScript%22%3A%22%2FApplications%2FAtom.app%2FContents%2FResources%2Fapp.asar%2Fsrc%2Finitialize-application-window.js%22%2C%22resourcePath%22%3A%22%2FApplications%2FAtom.app%2FContents%2FResources%2Fapp.asar%22%2C%22devMode%22%3Afalse%2C%22safeMode%22%3Afalse%2C%22profileStartup%22%3Afalse%2C%22clearWindowState%22%3Afalse%2C%22env%22%3A%7B%22TMPDIR%22%3A%22%2Fvar%2Ffolders%2F37%2Fnj5pdccs7d161n_ty00jdqj00000gn%2FT%2F%22%2C%22__CF_USER_TEXT_ENCODING%22%3A%220x1F5%3A0x0%3A0x0%22%2C%22HOME%22%3A%22%2FUsers%2Flocalhost%22%2C%22SHELL%22%3A%22%2Fbin%2Fbash%22%2C%22Apple_PubSub_Socket_Render%22%3A%22%2Fprivate%2Ftmp%2Fcom.apple.launchd.SHMoqWaqxQ%2FRender%22%2C%22SSH_AUTH_SOCK%22%3A%22%2Fprivate%2Ftmp%2Fcom.apple.launchd.3V1kbNY62p%2FListeners%22%2C%22PATH%22%3A%22%2Fusr%2Fbin%3A%2Fbin%3A%2Fusr%2Fsbin%3A%2Fsbin%22%2C%22LOGNAME%22%3A%22localhost%22%2C%22DISPLAY%22%3A%22%2Fprivate%2Ftmp%2Fcom.apple.launchd.AXwUwiN3q5%2Forg.macosforge.xquartz%3A0%22%2C%22XPC_SERVICE_NAME%22%3A%22com.github.atom.14500%22%2C%22USER%22%3A%22localhost%22%2C%22XPC_FLAGS%22%3A%220x0%22%2C%22GOOGLE_API_KEY%22%3A%22AIzaSyAQfxPJiounkhOjODEO5ZieffeBv6yft2Q%22%2C%22ATOM_HOME%22%3A%22%2FUsers%2Flocalhost%2F.atom%22%7D%2C%22appVersion%22%3A%221.14.3%22%2C%22atomHome%22%3A%22%2FUsers%2Flocalhost%2F.atom%22%2C%22shellLoadTime%22%3A438%7D:1

SyntaxError: Unexpected token L in JSON at position 0
    at Object.parse (native)
    at /packages/pros/lib/views/new-project.coffee:46:27)
    at HTMLButtonElement.dispatch (/packages/pros/node_modules/jquery/dist/jquery.js:4435:9)
    at HTMLButtonElement.elemData.handle (/packages/pros/node_modules/jquery/dist/jquery.js:4121:28)
```

### Commands

```
     -5:39.7.0 tree-view:show (atom-workspace.workspace.scrollbars-visible-when-scrolling.theme-one-dark-syntax.theme-one-dark-ui)
  2x -5:35.7.0 PROS:New-Project (input.hidden-input)
```

### Non-Core Packages

```
atom-beautify 0.29.17 
build 0.67.0 
busy 0.7.0 
file-icons 2.0.16 
linter 1.11.18 
platformio-ide-terminal 2.3.1 
pros 0.6.5 
tool-bar 1.0.1 
tool-bar-main 0.0.10 
```

edjubuh, 2017-02-27:
Thanks for the bug report!

Can you open Terminal and post the output of running `pros conduct lstemplate`? If everything appears fine, you can run `pros conduct new ./pros-test` to create a project in `./pros-test`

Reichenbachian, 2017-02-27:
Thank you for your quick response! That worked well.
[cid:E5B55669-B8BD-45C6-BBB3-77293D41200B@andover.edu]
I don’t have access to a cortex right now, so I can’t test it, but the project generation worked. I’ll tell you if it works when I next have access to a cortex.

—Alex Reichenbach

Reichenbachian, 2017-02-28:
Thank you for your previous help. It’s now hanging intermittently on “Stopping user code…” and “Asking for system information…” Do you have any further suggestions? I tried `pros upload` and `sudo pros upload`, but it still hung.

—Alex Reichenbach

HotelCalifornia, 2017-02-28:
Are you making sure to upload through the programming kit as seen [here](http://pros.cs.purdue.edu/known-issues/#linuxAA)?
If you are, can you tell me if this happens every time you go to flash code?

Reichenbachian, 2017-02-28:
The robot is connected to vex net, and, without changing the setup, I’m getting one of these errors every time I’ve tested.
[cid:BB9EA558-6833-4B70-BFD5-36F1EA054AEE@andover.edu][cid:996288F3-5631-4A2C-BED4-1629A2FA6974@andover.edu][cid:2CDC8C5C-E20C-4027-9188-249B09546518@andover.edu]

HotelCalifornia, 2017-02-28:
I'm not sure if you're trying to upload files or images, but we're not seeing any errors:

[image]

Reichenbachian, 2017-02-28:
Ah. I apologize. I’ll upload the output rather than the images then. I get one of these three scenarios each time I attempt to upload, without changing the setup at all.

Scenario 1
(anaconda3-4.1.0) Alexs-MBP:test localhost$ pros flash -f "/Users/localhost/Downloads/test"
Flashing /Users/localhost/Downloads/test/bin/output.bin to /dev/cu.usbmodem1411
Stopping user code... complete
Asking for system information... Asking for system information... complete
Cortex Microcontroller connected on /dev/cu.usbmodem1411
  Tether:   Serial w/ VEXnet 2.0 Keys
Joystick: F/W 4.25 w/ 4.84V
  Cortex: F/W 4.25 w/ 7.43V (Backup: 0.00V)
Sending to download channel... complete
Exposing bootloader... complete
Preparing bootloader…failed

Scenario 2
(anaconda3-4.1.0) Alexs-MBP:test localhost$ /Users/localhost/Desktop/Screen Shot 2017-02-28 at 3.37.54 PM.png

clear
pros flash -f "/Users/localhost/Downloads/test"
(anaconda3-4.1.0) Alexs-MBP:test localhost$
(anaconda3-4.1.0) Alexs-MBP:test localhost$ clear
(anaconda3-4.1.0) Alexs-MBP:test localhost$ pros flash -f "/Users/localhost/Downloads/test"
Flashing /Users/localhost/Downloads/test/bin/output.bin to /dev/cu.usbmodem1411
Stopping user code... complete
Asking for system information... ^C
Aborted!
(anaconda3-4.1.0) Alexs-MBP:test localhost$

Scenario 3
(anaconda3-4.1.0) Alexs-MBP:test localhost$ pros flash -f "/Users/localhost/Downloads/test"
Flashing /Users/localhost/Downloads/test/bin/output.bin to /dev/cu.usbmodem1411
Stopping user code... ^C
Aborted!
(anaconda3-4.1.0) Alexs-MBP:test localhost$

[https://camo.githubusercontent.com/0ee3f8feb1520d347a72aa963af287c26f0b202c/687474703a2f2f7075752e73682f756f614e512f643761303666636261342e706e67]<https://camo.githubusercontent.com/0ee3f8feb1520d347a72aa963af287c26f0b202c/687474703a2f2f7075752e73682f756f614e512f643761303666636261342e706e67>

Reichenbachian, 2017-02-28:
They all end with a control-C because they hang so long I’m forced to kill them.

On Feb 28, 2017, at 3:47 PM, George A Reichenbach <greichenbach@andover.edu<mailto:greichenbach@andover.edu>> wrote:

Ah. I apologize. I’ll upload the output rather than the images then. I get one of these three scenarios each time I attempt to upload, without changing the setup at all.

Scenario 1
(anaconda3-4.1.0) Alexs-MBP:test localhost$ pros flash -f "/Users/localhost/Downloads/test"
Flashing /Users/localhost/Downloads/test/bin/output.bin to /dev/cu.usbmodem1411
Stopping user code... complete
Asking for system information... Asking for system information... complete
Cortex Microcontroller connected on /dev/cu.usbmodem1411
  Tether:   Serial w/ VEXnet 2.0 Keys
Joystick: F/W 4.25 w/ 4.84V
  Cortex: F/W 4.25 w/ 7.43V (Backup: 0.00V)
Sending to download channel... complete
Exposing bootloader... complete
Preparing bootloader…failed

Scenario 2
(anaconda3-4.1.0) Alexs-MBP:test localhost$ /Users/localhost/Desktop/Screen Shot 2017-02-28 at 3.37.54 PM.png

clear
pros flash -f "/Users/localhost/Downloads/test"
(anaconda3-4.1.0) Alexs-MBP:test localhost$
(anaconda3-4.1.0) Alexs-MBP:test localhost$ clear
(anaconda3-4.1.0) Alexs-MBP:test localhost$ pros flash -f "/Users/localhost/Downloads/test"
Flashing /Users/localhost/Downloads/test/bin/output.bin to /dev/cu.usbmodem1411
Stopping user code... complete
Asking for system information... ^C
Aborted!
(anaconda3-4.1.0) Alexs-MBP:test localhost$

Scenario 3
(anaconda3-4.1.0) Alexs-MBP:test localhost$ pros flash -f "/Users/localhost/Downloads/test"
Flashing /Users/localhost/Downloads/test/bin/output.bin to /dev/cu.usbmodem1411
Stopping user code... ^C
Aborted!
(anaconda3-4.1.0) Alexs-MBP:test localhost$

[https://camo.githubusercontent.com/0ee3f8feb1520d347a72aa963af287c26f0b202c/687474703a2f2f7075752e73682f756f614e512f643761303666636261342e706e67]<https://camo.githubusercontent.com/0ee3f8feb1520d347a72aa963af287c26f0b202c/687474703a2f2f7075752e73682f756f614e512f643761303666636261342e706e67>

Reichenbachian, 2017-02-28:
I've removed VexNet and tried completely wired and still get the same errors.

Reichenbachian, 2017-02-28:
I can bypass Scenario 2 and 3 by plugging the robot in and quickly flashing the code before the robot has time to boot up.

Reichenbachian, 2017-02-28:
I have found that scenarios 2 and 3 are caused by a hang in `port.flush()`.

HotelCalifornia, 2017-02-28:
It seems the original pros-atom issue has been resolved, so caan you open an issue [purduesigbots/pros-cli](github.com/purduesigbots/pros-cli) for flasher issues?

closed 2017-02-28 by HotelCalifornia