purduesigbots/pros-atom #39: Hanging on Flash

Reichenbachian, 2017-02-28:
#### Expected Behavior:
I'm hoping for the program to flash to the VexNet.
#### Actual Behavior:
Instead, it is hanging on either 'Stopping user code' or 'Asking for system information'. Upon further inspection, I saw that both of these scenarios are caused by a hang in `port.flush()`.

I'm working with the OS-X version of atom and the most recent version of this repository.

Reichenbachian, 2017-02-28:
For further information:
On the times it receives data and hangs, the output of `received` is  b''
On the times it receives data and doesn't hang, the output of `received` is  b'\xaaU!\n\x04\x19\x04\x19s\x00\x00\x10\x00\t'

Reichenbachian, 2017-02-28:
I added `reset_cortex(port)` before the flush, and now it consistently gets past both  'Stopping user code' or 'Asking for system information'.
`def stop_user_code(port, ctx=proscli.utils.State()):
    reset_cortex(port)
    ....
`

edjubuh, 2017-03-17:
Alex, sorry for the lack of responses on our end. We've been working on making downloading more reliable 
on purduesigbots/pros-cli#27. If you have a chance, try seeing if you get better success on that branch. Run `pip install -e git+https://github.com/purduesigbots/pros-cli.git@flasher-tinkering#egg=pros-cli` to get this pre-production version. To get back out of it, just run `pip install pros-cli` and you'll be put back on the production version.

closed 2017-03-31 by edjubuh