purduesigbots/pros-atom #40: Failure on Preparing Bootloader

Reichenbachian, 2017-02-28:
#### Expected Behavior:
The function `prepare_bootloader` should prepare the bootloader, but it fails every time I've run it. The output is 'Preparing bootloader...failed'

Atom on MacOSX Sierra

jwalk511, 2017-03-01:
Not a fix, but a workaround. Download Java if you don't have it, and run `make upload` instead of `pros make upload`. This will directly call GNU make on the PROS makefiles, which will then compile your project and flash it to the cortex using Stephen's Uniflash (which is written in Java). Tests indicate that this is more stable than the Python flasher on macOS, though we aren't sure why.

HotelCalifornia, 2017-03-01:
We'd like to avoid forcing users to download more dependencies than are absolutely necessary, so we probably won't change the behavior of the upload button. We do have some flasher improvements in the pipeline over in purduesigbots/pros-cli#27, where I'll look into adding your suggested fix from #39.

edjubuh, 2017-04-01:
This should be fixed by purduesigbots/pros-cli@2a991c53

If you're still having issues with `pros upload` on version 2.5.0 or higher, feel free to reopen this issue.

closed 2017-04-01 by edjubuh