purduesigbots/pros-atom #41: Make failure: "stdlib.h: No such file..." in Arch Linux

benbroce3, 2017-03-28:
#### Actual Behavior:
`pros make` and `make` in terminal at the project root or building in Atom yields:

`
CC -I../include -I../src opcontrol.c
In file included from ../include/main.h:46:0,
                 from opcontrol.c:1:
../include/API.h:25:20: fatal error: stdlib.h: No such file or directory

                    ^
compilation terminated.
make[1]: *** [Makefile:44: ../bin/opcontrol.o] Error 1
make: *** [Makefile:44: src] Error 2
`

#### System information:
Operating System: (Manjaro [Arch] Linux)

Atom Version: (1.15.0)

Plugin Version: (0.6.9)

edjubuh, 2017-03-28:
How did you install PROS?

benbroce3, 2017-03-28:
Via the arch easy-install package. Would you recommend a re-install? Something else?

BottomNotch, 2017-03-29:
oops, the arch install script doesn't install newlib. run `sudo pacman -S arm-none-eabi-newlib` and you should be good to go.

benbroce3, 2017-03-29:
Thanks! You guys are awesome!

closed 2017-03-29 by benbroce3