purduesigbots/pros-atom #42: PROS package cannot load due to internal error

Octogonapus, 2017-05-05:
#### Expected Behavior:
Once Atom opens, all packages should load without errors.

#### Actual Behavior:
The PROS package throws a SyntaxError: https://hastebin.com/necapolite.md

#### Steps to reproduce:
I made a fresh Ubuntu 16.04 VM, ran pros-install_ubuntu.sh from 2.11.1: https://github.com/purduesigbots/pros/releases/tag/2.11.1

#### System information:
Operating System:  (Windows, OS X, Ubuntu)
Ubuntu 16.04
Atom Version: (Atom Version)
1.10.2
Plugin Version: (Plugin Version)
v0.7.0
#### Additional Information

#### Screenshots
@HotelCalifornia Asked to see if the package got loaded, it did not: https://github.com/purduesigbots/pros/releases/tag/2.11.1

Octogonapus, 2017-05-05:
I just installed on my laptop running Ubuntu 16.04 and got the same problem.

edjubuh, 2017-05-05:
Hi @Octogonapus, can you try updating your version of Atom? If this turns out to be the issue, I'll go ahead and get the installer script updated. https://github.com/atom/atom#debian-linux-ubuntu

Octogonapus, 2017-05-05:
Updating atom to latest (1.16.0) seems to have fixed it. The update required some addons to want updates as well, though (including linter which had dependencies).

edjubuh, 2017-05-05:
Weird. @HotelCalifornia you're done with your exams. Have fun :) (Cross referencing purduesigbots/pros#4)

HotelCalifornia, 2017-05-05:
Once we get the updated version of the install script from purduesigbots/pros#4 rolled into our releases, Ubuntu installs will fetch the latest version of atom from their website. If it still doesn't work after that, that means atom isn't updating their site properly

edjubuh, 2017-11-26:
I believe this has been resolved. Feel free to reopen if it's not

closed 2017-11-26 by edjubuh