purduesigbots/pros-atom #45: Uncaught Error: spawn ENOTDIR

brett-dun, 2017-07-07:
[Enter steps to reproduce:]

1. Switch from PROS GNU Make Default to PROS GNU Make Upload. Program gives errors that it did not previous give on a struct.
2. Press Save Program.

**Atom**: 1.18.0 x64
**Electron**: 1.3.15
**OS**: Mac OS X 10.12.1
**Thrown From**: [pros](https://github.com/purduesigbots/pros-atom) package 0.7.0

### Stack Trace

Uncaught Error: spawn ENOTDIR

```
At /Applications/Atom.app/Contents/Resources/app/src/buffered-process.js:392

Error: spawn ENOTDIR
    at exports._errnoException (util.js:1026:11)
    at ChildProcess.spawn (internal/child_process.js:313:11)
    at Object.exports.spawn (child_process.js:392:9)
    at BufferedProcess.spawn (/Applications/Atom.app/Contents/Resources/app/src/buffered-process.js:321:63)
    at BufferedProcess.start (/Applications/Atom.app/Contents/Resources/app/src/buffered-process.js:127:20)
    at new BufferedProcess (/Applications/Atom.app/Contents/Resources/app/src/buffered-process.js:112:18)
    at /packages/pros/lib/autocomplete/clang-provider.coffee:50:29
    at ClangProvider.module.exports.ClangProvider.codeCompletionAt (/packages/pros/lib/autocomplete/clang-provider.coffee:45:9)
    at ClangProvider.module.exports.ClangProvider.getSuggestions (/packages/pros/lib/autocomplete/clang-provider.coffee:37:8)
    at /Applications/Atom.app/Contents/Resources/app/node_modules/autocomplete-plus/lib/autocomplete-manager.js:337:60
    at Array.forEach (native)
    at AutocompleteManager.getSuggestionsFromProviders (/Applications/Atom.app/Contents/Resources/app/node_modules/autocomplete-plus/lib/autocomplete-manager.js:313:23)
    at AutocompleteManager.findSuggestions (/Applications/Atom.app/Contents/Resources/app/node_modules/autocomplete-plus/lib/autocomplete-manager.js:302:25)
```

### Commands

```
     -1:02.3.0 application:open-file (atom-pane.pane.active)
     -0:57.8.0 tree-view:show (atom-workspace.workspace.scrollbars-visible-when-scrolling.theme-one-dark-syntax.theme-one-dark-ui)
     -0:29.1.0 linter:next-error (atom-workspace.workspace.scrollbars-visible-when-scrolling.theme-one-dark-syntax.theme-one-dark-ui)
  2x -0:22 core:backspace (input.hidden-input)
  2x -0:20.1.0 core:save (input.hidden-input)
  3x -0:13.5.0 core:backspace (input.hidden-input)
     -0:12 core:save (input.hidden-input)
```

### Non-Core Packages

```
atom-beautify 0.30.3 
build 0.68.0 
busy 0.7.0 
file-icons 2.1.7 
linter 1.11.23 
platformio-ide-terminal 2.5.3 
pros 0.7.0 
tool-bar 1.1.0 
tool-bar-main 0.0.10 
```

HotelCalifornia, 2017-07-07:
Relevant line for investigation: https://github.com/purduesigbots/pros-atom/blob/master/lib/autocomplete/clang-provider.coffee#L43

HotelCalifornia, 2017-07-07:
@brett-dun would you mind providing a small code sample with which we can try to reproduce the error?

brett-dun, 2017-07-07:
```
enum SmartMotorPort {
  noMotor = -1, smart1 = 0, smart2, smart3, smart4, smart5, smart6, smart7, smart8, smart9, smart10
};
enum MotorPort {
  port1 = 1, port2, port3, port4, port5, port6, port7, port8, port9, port10
};
enum MotorType {
  normal = 0, high_speed, turbo
};
enum EncoderType {
  noEncoder = 0, internal, external
};
enum I2C_PORT {
  I2C_None = -1, I2C_1, I2C_2, I2C_3, I2C_4, I2C_5, I2C_6, I2C_7, I2C_8
};
enum DigitalPort {
  dgtl_1 = 1, dgtl_2, dgtl_3, dgtl_4, dgtl_5, dgtl_6, dgtl_7, dgtl_8, dgtl_9, dgtl_10, dgtl_11, dgtl_12, dgtl_13
};
enum AnalogPort {
  anlg_1 = 1, anlg_2, anlg_3, anlg_4, anlg_5, anlg_6, anlg_7, anlg_8
};

struct SmartMotor {
  MotorPort motorPort; //port that the motor is plugged into
  MotorType motorType; //type of motor
  I2C_PORT encoderPort;
  Encoder encoder;
  EncoderType encoderType;
  bool reversed;

  bool isPowerExpanderMotor; //whether or not it is plugged into a power expander
  SmartMotorPort slaves[5]; //slave motors that belong to this motor

  short slewRate; //rate at which motors speed is allowed to change per 20 ms

  float currentDraw; //amount of current that this motor draws from the port it is plugged into

  short currentSpeed; //speed that the cortex outputs
  short targetSpeed; //speed that the motor is targeted to be at

  float encoderRatio; //out:in
  int previousValue; //previous encoder value
  float velocity; //current velocity of the motor in RPMs
};
```

This is the code that would produce errors by the compiler and then cause Atom to hit the exception that it did.  I just started using PROS so I don't know if what I'm doing is allowed but there are no issues with this code when compiled with "make default" but once I change it to "make upload" I receive many errors.

HotelCalifornia, 2017-07-07:
Out of curiosity, what happens if you try to run `pros make flash` from within the project directory in a terminal?

brett-dun, 2017-07-07:
I get the message `make: *** No rule to make target 'flash'.  Stop.` but there is no exception with PROS.  When I select Make Upload, the exception does not occur every time.

HotelCalifornia, 2017-07-07:
Sorry, that was my mistake. I meant for the command to be either `pros mu` (runs make and then flash) or `pros flash`.

HotelCalifornia, 2017-07-26:
@brett-dun are you still having this issue?

brett-dun, 2017-07-31:
I have not been using the software much lately but when I have been using it, I did not run into any problems.

HotelCalifornia, 2017-07-31:
Alright! Thanks for letting us know.

I'm going to close this issue for now, but if you run into the same one again, please feel free to reopen. Otherwise, you can make a new issue.

closed 2017-07-31 by HotelCalifornia

aaronliu0130, 2019-02-24:
@HotelCalifornia Hello. I encountered the same problem after I closed the pros welcome screen, creating a new file "Untitled.cpp"(I deleted and saved it several times before as "Untitled" instead of "Untitled.cpp"), pasting into it some code(I have the "script" package) and then at least five of these popped out.
```
Applications/Atom.app/Contents/Resources/app/static/<embedded>:14
Hide Stack Trace
Error: spawn ENOTDIR
    at _errnoException (util.js:1024:11)
    at ChildProcess.spawn (internal/child_process.js:323:11)
    at Object.exports.spawn (child_process.js:514:9)
    at BufferedProcess.spawn (/Applications/Atom.app/Contents/Resources/app/static/<embedded>:14:1037304)
    at BufferedProcess.start (/Applications/Atom.app/Contents/Resources/app/static/<embedded>:14:1035479)
    at new BufferedProcess (/Applications/Atom.app/Contents/Resources/app/static/<embedded>:14:1035275)
    at /Users/Aaron/.atom/packages/pros/lib/autocomplete/clang-provider.coffee:50:25
    at new Promise (<anonymous>)
    at ClangProvider.module.exports.ClangProvider.codeCompletionAt (/Users/Aaron/.atom/packages/pros/lib/autocomplete/clang-provider.coffee:45:5)
    at ClangProvider.module.exports.ClangProvider.getSuggestions (/Users/Aaron/.atom/packages/pros/lib/autocomplete/clang-provider.coffee:37:8)
    at n.forEach.t (/Applications/Atom.app/Contents/Resources/app/static/<embedded>:11:647382)
    at Array.forEach (<anonymous>)
    at AutocompleteManager.getSuggestionsFromProviders (/Applications/Atom.app/Contents/Resources/app/static/<embedded>:11:646898)
    at AutocompleteManager.findSuggestions (/Applications/Atom.app/Contents/Resources/app/static/<embedded>:11:646636)
    at AutocompleteManager.requestNewSuggestions (/Applications/Atom.app/Contents/Resources/app/static/<embedded>:11:655166)
    at AutocompleteManager.showOrHideSuggestionListForBufferChanges (/Applications/Atom.app/Contents/Resources/app/static/<embedded>:11:656257)
    at Function.simpleDispatch (/Applications/Atom.app/Contents/Resources/app/static/<embedded>:11:1190583)
    at Emitter.emit (/Applications/Atom.app/Contents/Resources/app/static/<embedded>:11:1192024)
    at TextBuffer.emitDidChangeTextEvent (/Applications/Atom.app/Contents/Resources/app/static/<embedded>:11:503956)
    at TextBuffer.transact (/Applications/Atom.app/Contents/Resources/app/static/<embedded>:11:492131)
    at TextEditor.transact (/Applications/Atom.app/Contents/Resources/app/static/<embedded>:11:24215)
    at mergeIntersectingSelections (/Applications/Atom.app/Contents/Resources/app/static/<embedded>:11:16150)
    at TextEditor.mergeSelections (/Applications/Atom.app/Contents/Resources/app/static/<embedded>:11:39529)
    at TextEditor.mergeIntersectingSelections (/Applications/Atom.app/Contents/Resources/app/static/<embedded>:11:39038)
    at TextEditor.mutateSelectedText (/Applications/Atom.app/Contents/Resources/app/static/<embedded>:11:16113)
    at TextEditor.insertText (/Applications/Atom.app/Contents/Resources/app/static/<embedded>:11:15662)
    at TextEditor.e.(anonymous function) [as insertText] (/Applications/Atom.app/Contents/Resources/app/static/<embedded>:11:259206)
    at TextEditorComponent.didTextInput (/Applications/Atom.app/Contents/Resources/app/static/<embedded>:11:93878)
```
[image]

aaronliu0130, 2019-02-24:
I know this might be not related to atom at all, but I still need an answer.

aaronliu0130, 2019-02-24:
[image]

HotelCalifornia, 2019-02-24:
@aaronliu0130 I'm confused. In your screenshot, it doesn't look like you have a PROS project open.

What were you doing when you ran into this error?

aaronliu0130, 2019-02-28:
I just closed the pros welcome screen, created a new file(without an extension), saved it multiple times, added the cpp extension, pasted two times cpp code in it and saving each time. From now on, I don't really remember. Then, I browsed packages, searched for build, browsed install, searched for build again, and returned to the cpp file.
Again, this might not be related to pros at all.

aaronliu0130, 2019-02-28:
I saw something in common with the first comment: we were both dealing with something about the c language.

aaronliu0130, 2019-02-28:
I don't know how I ran in it, it seemed completely random