purduesigbots/pros-atom #47: Key files empty in new project

Riley3116C, 2017-07-11:
#### Expected Behavior:   New project is created, with init.c , main.h and API.h all ready to use

#### Actual Behavior:      New project is created, with init.c , main.h and API.h all blank

#### Steps to reproduce:    Just create new projects I guess?

#### System information:
Operating System:  (Windows, OS X, Ubuntu) Windows 7

Atom Version: (Atom Version) 1.15.0

Plugin Version: (Plugin Version) Latest

#### Additional Information   Was fixed after recreating the project. May be good to look into it anyway though

#### Screenshots   None. Sorry. I contacted the guys in the Discord for help on the issue

HotelCalifornia, 2017-07-11:
This doesn't seem to be the first report we've had of this particular issue occurring on Windows 7, but we still need to be able to reproduce it reliably in order to figure out what's going on.

HotelCalifornia, 2017-07-11:
If anyone has experienced this issue, please let us know

HotelCalifornia, 2017-07-11:
From our previous correspondence with a user reporting this issue:

> I also have found a problem with Windows 7 installations of PROS for Atom.  The “Create a new PROS project” command will create a project in the <UserName>/AppData/Local/atom/app-1.18.0 folder, and the project is fully populated.  However, in the editor the auto.c, init.c, and opcontrol.c files are all empty.  For example, if I open the src folder in the Project tree, and click on “auto.c”, the editor tab is empty, and the window title is “auto.c – src – TestProject – Atom”.
>If I then right-click on the project in the project tree and select “Open in New Window”, a new Atom window will open with the title “auto.c – C:\Users\<UserName>\AppData\Local\atom\app-1.18.0\TestProject – Atom”.  In this window I can successfully build the project.

fraxker, 2017-08-06:
I have run into this problem now on 2 different computers, both running windows 10. Just retryed creating the file and by changing the directory from the PROS program folder to a different location fixed it. May have to do with restricted folder access?

HotelCalifornia, 2017-08-06:
> May have to do with restricted folder access?
Yeah, you shouldn't be putting your projects in the PROS program folder- program folders restrict write access to administrators only by default.

fraxker, 2017-08-06:
For some reason, that is the directory it defaulted to, don't know if that was something I did or not

closed 2017-12-10 by edjubuh