purduesigbots/pros-atom #49: Incorrect project hierarchy for new projects using 2.12

bradamiller, 2017-07-25:
#### Expected Behavior:
Creating a new PROS project should have a firmware directory with all the supporting files

#### Actual Behavior:
There is a nested firmware directory inside the correct firmware directory that contains cortex.ld, STM32F10x.ld, and uniflash.jar. The outer firmware directory has libpros.a.

#### Steps to reproduce:
Create a PROS project using the PROS menu in ATOM.

#### System information:
macOS 10.12.6

Atom Version:

Plugin Version (PROS version):
2.12

#### Additional Information

#### Screenshots
[image]

edjubuh, 2017-07-25:
Thanks for noticing this bug!

I looked at our template and looks like we had done that nesting by accident. I've re-uploaded the kernel template. You can force download the 2.12.0 kernel template by running `pros conduct download 2.12.0 -nc`.

I would recommend simply moving the nested files back out of the directory rather than trying to recreate the project.

Feel free to close this if this solved your issue.

bradamiller, 2017-07-25:
It didn’t work. Can I use the PROS Conductor to revert back to a previous version, then up to 2.12?

new-host-9:PROS_2.0 bradmiller$ pros conduct download 2.12.0 -nc
Fetching online listing to verify available templates.
No templates were found with the name 2.12.0
Aborted!

edjubuh, 2017-07-25:
Apologies, `pros conduct download kernel 2.12.0 -nc` (or, while it's the latest version, `pros conduct download -nc`)

HotelCalifornia, 2017-07-26:
@bradamiller did the above revised command work to solve your problem?

bradamiller, 2017-07-26:
Thanks for checking. 

It is working now and I’m using the conductor to update all of our laptops to the updated 2.12 that Elliot put up. 

Also a few quick questions. The conductor lets me open and create projects. That’s duplicated on the pros menu in atom. Is one of those method “preferred“?  And it looks like in the conductor I can add libraries to the project. Do those contain the header file and the archive and if so what’s the convention for the directory structure for that to happen? I have a driving library similar to what I did for FRC. 

Last question, if I create cpp files with classes do those work?

Thanks
Brad

HotelCalifornia, 2017-07-26:
Neither method is preferred by us, they both exist to provide different ways to do the same thing (that is, running `pros conduct new`).

Libraries can be made and distributed, like we do with [libblrs](https://github.com/purduesigbots/libblrs/releases). Currently, we don't have any explanations on how to do this, but rest assured I will be writing up a thorough tutorial tonight.

With regard to your question about C++, please see [this post](https://www.vexforum.com/index.php/28287-new-and-delete-keywords-pros/p1#p230222) on the VEX forum.

I'm going to close this issue for now, if you have any more questions, feel free to shoot us an email: pros_development@cs.purdue.edu

closed 2017-07-26 by HotelCalifornia