purduesigbots/pros-atom #50: No Text on PROS CLI

extension5525A, 2017-08-10:
#### Expected Behavior: Text to come up indicating that a project was being loaded to robot and and text

#### Actual Behavior: No text comes up on PROS CLI and doesn't even respond to a definitely wrong command. Basically me putting in a bunch or random letters.

#### Steps to reproduce: Any interaction that should produce a text on the PROS CLI command line like thing did nothing and no text appeared.

#### System information: Widows 10
Operating System:  (Windows, OS X, Ubuntu)

PROS Version: (run `pros --version`) Not sure but downloaded it within the past week from the website.

#### Additional Information I am using atom. I restated atom and my computer but the problem persisted

#### Screenshots/Output Dumps/Stack Traces
[image]

HotelCalifornia, 2017-08-10:
Can you be more specific with respect to "any interaction"? What interactions did you try?

Also, can you open up a command prompt (type "cmd" into the search bar and it should be the first result) and run `pros --version`? The reason we ask users to do so is not only to report the version, but also to check that the PROS CLI has been properly installed.

In your screenshot, it looks like you have some package updates-- can you try updating them and see if that fixes your issue?

extension5525A, 2017-08-10:
Well I tried to upload the project, type in random words into the line like when you click the keyboard button on the top left. It usually tells me that is a false command but nothing happened. I also tried to open the serial window and that changed nothing also. It is version 2.5.1

extension5525A, 2017-08-10:
The updates is what fixed it thanks. Its resolved now.

closed 2017-08-10 by extension5525A