purduesigbots/pros-atom #51: Uncaught Error: ENOENT: no such file or directory, stat 'arm-none-eabi-g  '

NEMS05, 2017-09-27:
[Enter steps to reproduce:]

1. ...
2. ...

**Atom**: 1.20.1 x64
**Electron**: 1.6.9
**OS**: Mac OS X 10.12.6
**Thrown From**: [pros](https://github.com/purduesigbots/pros-atom) package 0.7.1

### Stack Trace

Uncaught Error: ENOENT: no such file or directory, stat 'arm-none-eabi-g  '

```
At fs.js:912

Error: ENOENT: no such file or directory, stat 'arm-none-eabi-g  '
    at fs.statSync (fs.js:912:18)
    at Object.fs.statSync (ELECTRON_ASAR.js:280:16)
    at /packages/pros/lib/lint.coffee:50:38
    at /packages/pros/node_modules/command-exists/lib/command-exists.js:53:23
    at ChildProcess.exithandler (child_process.js:230:5)
    at emitTwo (events.js:106:13)
    at ChildProcess.emit (events.js:191:7)
    at maybeClose (internal/child_process.js:885:16)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:226:5)
```

### Commands

```
     -6:17.2.0 application:reopen-project (input.hidden-input)
     -6:17 tree-view:show (atom-workspace.workspace.scrollbars-visible-when-scrolling.theme-one-dark-syntax.theme-one-dark-ui)
  2x -0:09.8.0 core:backspace (input.hidden-input)
```

### Non-Core Packages

```
atom-beautify 0.30.5 
build 0.68.0 
busy 0.7.0 
busy-signal 1.4.3 
file-icons 2.1.12 
git-plus 7.9.3 
import 1.7.0 
intentions 1.1.5 
linter 2.2.0 
linter-ui-default 1.2.4 
platformio-ide-terminal 2.7.0 
pros 0.7.1 
tool-bar 1.1.0 
tool-bar-main 0.0.10 
```

HotelCalifornia, 2017-10-05:
Looks as if the grepping for arm-none-eabi-g++ is getting broken again

edjubuh, 2017-10-23:
@NEMS05 do you have steps to reproduce?

HotelCalifornia, 2018-07-06:
closing due to lack of activity

closed 2018-07-06 by HotelCalifornia