purduesigbots/pros-atom #56: Unable to Find CLI/Kernel

Highway-Man, 2017-11-26:
#### Expected Behavior:
Atom finds kernel version and can initialize new projects and do other normal things without errors

#### Actual Behavior:
Atom throws "Uncaught TypeError: Invalid Version: undefined" and cannot find CLI version

#### Steps to reproduce:
Install latest release without atom previously installed and open PROS Editor

#### System information:
Operating System (e.g. Windows, OS X, Ubuntu):
Windows 10 32-bit

Atom Version:
1.21.2

Plugin Version (PROS version):
0.7.4

#### Additional Information
Also, if I try to create a new project, register, or upgrade an existing one, atom says "Error obtaining list of kernels downloaded. STDOUT: ERR: usage: pros [-h] [--version] {create, upgrade, fetch, config} ... pros: error invalid choice: 'conduct' (choose from 'create', 'upgrade', 'fetch', 'config')

#### Screenshots
[image]

edjubuh, 2017-11-26:
What happens when you run `pros --version` from command prompt or Powershell?

Highway-Man, 2017-11-26:
Wow, that was fast!
It returns 0.2.0

edjubuh, 2017-11-26:
Hmmm... Can you restart PROS Editor completely? Close it out, wait a few seconds, and start it up again.

Highway-Man, 2017-11-26:
no change

edjubuh, 2017-11-26:
Can you run `where pros`? You have a really old version of the PROS CLI there.

Highway-Man, 2017-11-26:
C:\Program Files\PROS\cli\pros.exe

presumably from way back when . . . I've been putting off switching to Atom. So the installer doesn't update it to the newest version? I guess I have to use the conductor process for upgrading described on your website

edjubuh, 2017-11-26:
0.2.0 predates conductor. Do you have any other versions of PROS installed?

Highway-Man, 2017-11-26:
I still have the eclipse-based IDE. I don't think I have any other versions of the cli anywhere

edjubuh, 2017-11-26:
We don't really support eclipse and newer versions side by side. You'll need to trick either atom or eclipse into finding the right version first. The simpler solution may just be to uninstall the older version. Comment back if you need help with either solution.

Highway-Man, 2017-11-26:
OK, that's fine. I'll uninstall everything and rerun the new one. Is there anything I need to delete/change manually? I was mostly just going to wait to uninstall eclipse until after I saw that this was working, but I guess that backfired

btw, you might want to update this. I guess it already says "possible" but I took it to mean the uninstaller thing was the only issue
https://pros.cs.purdue.edu/getting-started/windows/#installing-alongside-pros-for-eclipse

Highway-Man, 2017-11-26:
Uninstalling both the old and new and rerunning the latest installer seems to have fixed it

Highway-Man, 2017-11-26:
. . . Or not
So now it finds the CLI version 2.6.1 and there's no red error box. I can create a new project without issues. But when I run build: trigger I get the output below.

process_begin: CreateProcess(NULL, true, ...) failed.
make (e=2): The system cannot find the file specified.
make: *** [_force_look] Error 2

edjubuh, 2017-11-27:
What's the environment variable PROS_TOOLCHAIN set to?

Highway-Man, 2017-11-28:
It doesn't exist. There's no entry for it

edjubuh, 2017-11-28:
Can you either manually set PROS_TOOLCHAIN to `C:\Program Files\PROS\toolchain\usr` or run the PROS installer and choose the repair option. I see the environment variables occasionally not getting set, but I haven't been able to reproduce the issue.

Highway-Man, 2017-11-29:
Running repair didn't fix it. So I set it manually. Now when I run build, Atom throws a red popup saying spawn unknown and the make -j1 pane opens with no text in it. Running build again causes Atom to give another popup saying Status: 'Build: PROS GNU Make: default' is already set

closed 2018-12-20 by HotelCalifornia