purduesigbots/pros-atom #57: Error: No microcontrollers were found. Please plug in a cortex or manually specify a serial port.'

safsom, 2017-11-30:
#### Expected Behavior:

#### Actual Behavior:

#### Steps to reproduce:

#### System information:
Operating System (e.g. Windows, OS X, Ubuntu):

Atom Version:

Plugin Version (PROS version):

#### Additional Information

#### Screenshots

HotelCalifornia, 2017-11-30:
VEX doesn't provide drivers for Linux-based operating systems, so make sure you are following [the instructions here](https://pros.cs.purdue.edu/known-issues/#linuxAA) when connecting your cortex to your computer.

josseanyamil, 2018-04-10:
I have the same problem. I'm using an old Programming module (before 2012) on a Mac; I have the PROS Editor with Java and Python 3 installed. I make sure that everything is connected to the computer but the same error shows up "No microcontrollers were found". I can't find any help online.

ryanbloom, 2018-12-13:
I'm also having this problem, also with a Mac and a pre-2012 programming module. Is PROS only compatible with the new programming module?