purduesigbots/pros-atom #58: how to open autocomeple?

3038922, 2017-12-06:
Excuse me
I am a student,i come form china .my englist is poor.
I can't use autocomeple.
How to set up?autocomeple flags?

3038922, 2017-12-06:
[image]

3038922, 2017-12-06:
I want to enter motoSet(1,127) ; ,but  only main and mark.

edjubuh, 2017-12-06:
Autocomplete gives you suggestions once you type 3 or more characters. Do you see `motorSet` once you type `mot`? If so, I can give you instructions to decrease the number of characters required so you see suggestions starting at 1 or 2 characters.

3038922, 2017-12-06:
[image]
I've been reloading many times. I think it's my system setting problem.

3038922, 2017-12-06:
[image]

edjubuh, 2017-12-06:
OK, can you run Command Prompt and type `echo %PATH%`? You should see something like `C:\Program Files\PROS\llvm\bin` in the list.

3038922, 2017-12-06:
[image]

3038922, 2017-12-06:
[image]

edjubuh, 2017-12-06:
Ah, it looks like you're not using our distributed version of PROS Editor which includes the necessary dependencies. We fully support using Atom the way you have installed, but you'll need to install clang [here](http://releases.llvm.org/5.0.0/LLVM-5.0.0-win64.exe).

3038922, 2017-12-06:
I'm going to use ATOM to learn C++, so I didn't install PROS EDITOR.
[image]

3038922, 2017-12-06:
I installed EDITOR in my system to make mistakes. I can't use GCC-RUN and LINTER-GCC  as a compiler.

3038922, 2017-12-06:
[image]
[image]

3038922, 2017-12-06:
[image]

edjubuh, 2017-12-06:
> I can't use GCC-RUN and LINTER-GCC as a compiler.
We don't support those plugins. Using the PROS plugin will get you the ability to lint and build your projects.

I'm not sure how you have PROS installed anymore. Do you still not have the PROS Editor installed? If so, please install clang as mentioned [above](https://github.com/purduesigbots/pros-atom/issues/58#issuecomment-349721455).

3038922, 2017-12-07:
thank you .I solved the problem of AUTOCOMEPLE after I reinstalled.There are many junior high school students in China like PROS.Our teachers don't understand it very well. We need to study ourselves completely.We have finished the teaching of the official website, and hope to learn more.

3038922, 2017-12-07:
ouer team number is 7995A.

edjubuh, 2017-12-07:
> There are many junior high school students in China like PROS.

Glad to hear that PROS is being used overseas! We're always looking to give students ready-to-use STEM skills!

> Our teachers don't understand it very well. We need to study ourselves completely.We have finished the teaching of the official website, and hope to learn more.

Feel free to submit suggestions/open issues on [purduesigbots/pros-docs](https://github.com/purduesigbots/pros-docs) if you'd like anything clarified.

closed 2017-12-07 by edjubuh