purduesigbots/pros-atom #59: How do you start automatic at the console?

3038922, 2017-12-10:
Excuse me, I am a novice.
I used easyc before.
I have many many question......
1 How do you start automatic at the console?
2 Can you provide an online test?like easyc.
Can display all sensor data in real time.
3 Please provide examples of some of the TASK functions.
4 I hope to provide a complete example of the VEX program for learning.
The program in the tutorial is too simple.
Thank you !!Thank you very much.

3038922, 2017-12-10:
Do you have a Chinese team? I want to have a Chinese explanation...
~.~

edjubuh, 2017-12-10:
1) PROS Editor doesn't have this functionality. You can use `pros mut` from the terminal to make, upload, and open up the terminal.

2) We're developing JINX as a solution to this, but it's not currently ready for general use.

3) Have you taken a look at our [Tasks](https://pros.cs.purdue.edu/tutorials/tasks/) tutorial?

3038922, 2017-12-11:
1I want to know what is the console command to start the auto race?
3 Yes, but it's hard to understand for a high school student.
Although PROS is very difficult for us to get started, I like the style of ATOM very much.
^^

closed 2018-07-06 by HotelCalifornia