purduesigbots/pros-atom #61: How to update ATOM?

3038922, 2018-01-03:
[image]
[image]
[image]
Maybe it's normal to upgrade ATOM LINTER.

HotelCalifornia, 2018-01-03:
It looks like you're using the PROS Editor, which we maintain separately from the main Atom repository. In other words, we have the version pinned at 1.21.2 until we can merge in upstream changes.

3038922, 2018-01-03:
Thank you.
But I upgraded LINTER to make it impossible to display it normally.

closed 2018-07-06 by HotelCalifornia