purduesigbots/pros-atom #65: Minor detail - PROS project icon

KZeronimo, 2018-07-28:
Very minor detail - more curious than anything (tried to correct with providing icon properties override in styles.less no luck)

#### Expected Behavior:
Project Icon for pros project sized appropriately

#### Actual Behavior:
pros project icon too large

#### Steps to reproduce:
Install Atom 
Install PROS utilizing Windows Installer (uncheck PROS editor)
Install PROS plugin `apm install file-icons linter tool-bar tool-bar-main busy build platformio-ide-terminal pros`
Create PROS project

#### System information:
Operating System (e.g. Windows, OS X, Ubuntu):
Window 10 1803

Atom Version:
1.28.2 x64
Plugin Version (PROS version):
0.7.4

#### Additional Information

#### Screenshots
[image]

closed 2018-07-30 by edjubuh

KZeronimo, 2018-07-30:
@edjubuh - I see file icon hack from purdueos.less was removed and I also see the `.pros-icon` is defined as an svg trace and that seems to be what is painting in the original screenshot I submitted - v0.7.5 of the plugin is still producing the original screenshot - is that the expected behavior?

I was hacking around and if I remove `.pros-icon` from purdueos.less then I observe the following which is what I would have expected (the PROS icons handled by [file-icons](https://github.com/file-icons/source/blob/master/charmap.md) `\EAAD`)

[image]

I'm not sure what the project pane is supposed to look like wrt to the icon for the pros file extension and I'm confident my hack is not the solution - just wanted to give you some more info - I know you guys have bigger fish to fry with v3 :)

Keep up the good work

edjubuh, 2018-07-30:
Do I need to remove more than what I did in #66?

PR #66 (2018-07-30): Remove file-icon hack
Upstream has the PROS icon now

Closes #65
commit 8d95fce7 (2018-07-30): Remove file-icon hack

Upstream has the PROS icon now

Closes #65