purduesigbots/pros-atom #68: Deprecation of path.dirname

aaronliu0130, 2018-10-07:
#### Expected Behavior:
Non
#### Actual Behavior:
Deprecated calls
pros (1 deprecation)
Check for UpdateDisable Package
Argument to path.dirname must be a string
Report Issue
Object.dirname - /Applications/Atom.app/Contents/Resources/app.asar/src/electron-shims.js:9:10
NewProjectModal.initialize - /Users/Aaron/.atom/packages/pros/lib/views/new-project.coffee:118:36
NewProjectModal.constructor - /Users/Aaron/.atom/packages/pros/node_modules/space-pen/lib/space-pen.js:184:25
new NewProjectModal - /Users/Aaron/.atom/packages/pros/lib/views/new-project.coffee:26:52
HTMLElement.PROS:New-Project - /Users/Aaron/.atom/packages/pros/lib/main.coffee:84:22
CommandRegistry.handleCommandEvent - <embedded>:24144:49

#### Steps to reproduce:
Don't know. Just ran int this "deprecation error".
#### System information:
Operating System (e.g. Windows, OS X, Ubuntu):
macOS Mojave Version 10.14 (18A391)
Atom Version:
1.31.2
Plugin Version (PROS version):
0.7.5
#### Additional Information

#### Screenshots
[image]
[image]
[image]

aaronliu0130, 2018-10-07:
When I pressed "Check for Update" in the Deprecation Cup, it said everything was up to date.
[image]

HotelCalifornia, 2018-12-20:
> why isn't it fixed on mac?

because it looks like this is being thrown from a different place than in #36. Either way, this can be safely ignored (and we're developing a new version of the atom plugin anyway, at purduesigbots/pros-atom3)

closed 2018-12-20 by HotelCalifornia