purduesigbots/pros-atom #73: Rename pros package to pros2

edjubuh, 2019-04-29:
Should be renamed to something else since the current name ("pros") makes it seem like this is the latest version.

HotelCalifornia, 2019-05-01:
https://flight-manual.atom.io/behind-atom/sections/maintaining-your-packages/#rename-your-package

HotelCalifornia, 2019-05-01:
may also want to rename/archive this repo as well, if we're not going to do any more work here (which is likely)

Samuel12321, 2019-05-10:
#71

HotelCalifornia, 2019-05-10:
@Samuel12321 yeah, potentially. Currently pros-bootstrapper@0.12 will install all the packages needed for the PROS 3 ecosystem, but it's not documented that well partially by design-- in general the only people who should be installing that package manually are those who know what they're doing (or following the instructions I linked to you in a previous comment)